eclipse-zenoh/zenoh-c #20: Compatibility with atolab/zhe?

gavanderhoorn, 2021-02-16:
What's the compatibility between the ZHe implementation here and the (previous?) version still hosted at [atolab/zhe](https://github.com/atolab/zhe)?

I'm trying to see whether it'd be possible to build ZHe on a platform (non-posix) which doesn't support Rust (or at least not directly), but IIUC, `zenoh-c` requires Rust as it wraps the Rust implementation in [eclipse-zenoh/zenoh](https://github.com/eclipse-zenoh/zenoh). `atolab/zhe` is plain C, which seems more feasible at the moment. Whether it'd be usable depends on the compatibility with the current Rust based implementation.

Admittedly, I may be over-complicating things: if building a static library from the Rust implementation is not that difficult, I could see whether that´d be possible, but I expect that'd need building a Rust compiler first (which is not necessarily feasible).

PS: this may be an RTFM, if so, please point me to TFM :)

JEnoch, 2021-02-16:
Hi @gavanderhoorn ,

The equivalent of `zhe` is now [eclipse-zenoh/zenoh-pico](https://github.com/eclipse-zenoh/zenoh-pico).  
It's plain C and protocol-compatible with the latest Rust-based [eclipse-zenoh/zenoh](https://github.com/eclipse-zenoh/zenoh) versions.

gavanderhoorn, 2021-02-16:
Ah, that's great to hear. Thanks for that.

Perhaps @eboasson could add a note to `atolab/zhe` to point people to `zenoh-pico`?

gavanderhoorn, 2021-02-16:
I'll close this then. Thanks again.

closed 2021-02-16 by gavanderhoorn

eboasson, 2021-02-16:
@gavanderhoorn thanks for the suggestion!

gavanderhoorn, 2021-02-16:
Somewhat off-topic, but: it seems `zenoh-pico` uses threads, whereas `atolab/zhe` does not. The single-threaded nature of `atolab/zhe` made for some really easy porting.