eclipse-zenoh/zenoh-c #26: make fails : use of unstable library feature 'renamed_spin_loop'

vortex314, 2021-07-05:
Compiling spinning_top v0.2.4
error[E0658]: use of unstable library feature 'renamed_spin_loop'
  --> /home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/spinning_top-0.2.4/src/spinlock.rs:57:17
   |
57 |                 hint::spin_loop();
   |                 ^^^^^^^^^^^^^^^
   |
   = note: see issue #55002 <https://github.com/rust-lang/rust/issues/55002> for more information

error: aborting due to previous error

For more information about this error, try `rustc --explain E0658`.
error: could not compile `spinning_top`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
make: *** [Makefile:55: target//release/libzenohc.so] Error 101

vortex314, 2021-07-05:
Ok, seems linked to use of unstable versions. 
Installing rustup...
´´´
lieven@pcpav2:~/workspace/zenoh-c$ sudo snap install rustup
error: This revision of snap "rustup" was published using classic confinement and thus may perform
       arbitrary system changes outside of the security sandbox that snaps are usually confined to,
       which may put your system at risk.

       If you understand and want to proceed repeat the command including --classic.
´´´
gives confidence to the  unexpected user.. ;-)

vortex314, 2021-07-05:
After installing the dependecies,
```
 2263  sudo snap install rustup
 2264  rustup toolchain install nightly
 2265  sudo snap install rustup --classic
 2266  rustup toolchain install nightly 
 2267  export PATH=$PATH:/snap/bin
 2268  rustup toolchain install nightly 
 2269  rustup default nightly
 2270  cargo install cbindgen
 2271  export PATh=$PATH:/home/lieven/.cargo/bin
 2272  make
```
still the same issue : ```
  Compiling atty v0.2.14
   Compiling pnet_sys v0.28.0
error[E0658]: use of unstable library feature 'renamed_spin_loop'
  --> /home/lieven/.cargo/registry/src/github.com-1ecc6299db9ec823/spinning_top-0.2.4/src/spinlock.rs:57:17
   |
57 |                 hint::spin_loop();
   |                 ^^^^^^^^^^^^^^^
   |
   = note: see issue #55002 <https://github.com/rust-lang/rust/issues/55002> for more information

error: aborting due to previous error

For more information about this error, try `rustc --explain E0658`.
error: could not compile `spinning_top`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
make: *** [Makefile:55: target//release/libzenohc.so] Error 101
```

Mallets, 2021-07-06:
zenoh requires a recent Rust version. The Rust version shipped with Ubuntu is quite old and it gives some headache when compiling (see https://github.com/eclipse-zenoh/zenoh/pull/94). 

Please install the latest Rust version following [these instructions](https://www.rust-lang.org/tools/install).

vortex314, 2021-07-06:
Ok, will try that.

vortex314, 2021-07-06:
Thanks for your help @Mallets ,  compile issue solved !

closed 2021-07-06 by vortex314