eclipse-zenoh/zenoh-c #27: creating loop in examples generates core dump

vortex314, 2021-07-14:
I took the example zn_pub and adpated to create a loop around zn_open and zn_close.
However after a second loop it crashes :+1: 
```
(gdb) run
Starting program: /home/lieven/workspace/zenoh-proxy/build/zn_pubsub 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Openning session...
[New Thread 0x7ffff74df700 (LWP 7629)]
[New Thread 0x7ffff72de700 (LWP 7630)]
[New Thread 0x7ffff70dd700 (LWP 7631)]
[New Thread 0x7ffff6edc700 (LWP 7632)]
[New Thread 0x7ffff6cdb700 (LWP 7633)]
[New Thread 0x7ffff6ada700 (LWP 7634)]
[New Thread 0x7ffff68d9700 (LWP 7635)]
[New Thread 0x7ffff66d8700 (LWP 7636)]
[New Thread 0x7ffff64d7700 (LWP 7637)]
[New Thread 0x7ffff62d6700 (LWP 7638)]
[New Thread 0x7ffff60d5700 (LWP 7639)]
[New Thread 0x7ffff5ed4700 (LWP 7640)]
[New Thread 0x7ffff5cd3700 (LWP 7641)]
[New Thread 0x7ffff5ad2700 (LWP 7642)]
[New Thread 0x7ffff58d1700 (LWP 7643)]
[New Thread 0x7ffff56d0700 (LWP 7644)]
[New Thread 0x7ffff54cf700 (LWP 7645)]
[New Thread 0x7ffff52ce700 (LWP 7646)]
[New Thread 0x7ffff50cd700 (LWP 7647)]
[New Thread 0x7ffff4ecc700 (LWP 7648)]
[New Thread 0x7ffff4ccb700 (LWP 7649)]
[New Thread 0x7ffff4aca700 (LWP 7650)]
Declaring Resource '/demo/example/zenoh-c-pub' => RId 1
Declaring Publisher on 1
Writing Data ('1': '[   0] Pub from C!')...
Writing Data ('1': '[   1] Pub from C!')...
Openning session...

Thread 1 "zn_pubsub" received signal SIGSEGV, Segmentation fault.
0x00007ffff79aed18 in <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll () from /usr/local/lib/libzenohc.so
(gdb) bt
#0  0x00007ffff79aed18 in <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll ()
   from /usr/local/lib/libzenohc.so
#1  0x00007ffff791206d in <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll ()
   from /usr/local/lib/libzenohc.so
#2  0x00007ffff793fea0 in std::thread::local::LocalKey<T>::with () from /usr/local/lib/libzenohc.so
#3  0x00007ffff7916618 in <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll ()
   from /usr/local/lib/libzenohc.so
#4  0x00007ffff7938506 in async_io::driver::block_on () from /usr/local/lib/libzenohc.so
#5  0x00007ffff794269c in std::thread::local::LocalKey<T>::with () from /usr/local/lib/libzenohc.so
#6  0x00007ffff79408fa in std::thread::local::LocalKey<T>::with () from /usr/local/lib/libzenohc.so
#7  0x00007ffff794323d in std::thread::local::LocalKey<T>::with () from /usr/local/lib/libzenohc.so
#8  0x00007ffff7927a14 in async_std::task::builder::Builder::blocking () from /usr/local/lib/libzenohc.so
#9  0x00007ffff794f33e in zn_open () from /usr/local/lib/libzenohc.so
#10 0x000055555555542c in main (argc=1, argv=0x7fffffffd818) at /home/lieven/workspace/zenoh-proxy/src/zn_pubsub.c:23
(gdb) 
```
source
```
int main(int argc, char **argv) {
  z_init_logger();

  char *uri = "/demo/example/zenoh-c-pub";
  if (argc > 1) {
    uri = argv[1];
  }
  char *value = "Pub from C!";
  if (argc > 2) {
    value = argv[2];
  }
  zn_properties_t *config = zn_config_default();
  if (argc > 3) {
    zn_properties_insert(config, ZN_CONFIG_PEER_KEY, z_string_make(argv[3]));
  }
  while (true) {
    printf("Openning session...\n");
    zn_session_t *s = zn_open(config);
    if (s == 0) {
      printf("Unable to open session!\n");
      exit(-1);
    }

    printf("Declaring Resource '%s'", uri);
    unsigned long rid = zn_declare_resource(s, zn_rname(uri));
    printf(" => RId %lu\n", rid);
    zn_reskey_t reskey = zn_rid(rid);

    printf("Declaring Publisher on %lu\n", rid);
    zn_publisher_t *pub = zn_declare_publisher(s, reskey);
    if (pub == 0) {
      printf("Unable to declare publisher.\n");
      exit(-1);
    }

    char buf[256];
    for (int idx = 0; idx<2; ++idx) {
      sprintf(buf, "[%4d] %s", idx, value);
      printf("Writing Data ('%lu': '%s')...\n", rid, buf);
      zn_write(s, reskey, (const uint8_t *)buf, strlen(buf));
    }
    zn_undeclare_publisher(pub);
    zn_close(s);
  };
  return 0;
}
```
Independent if zenohd is running in the background or not.

vortex314, 2021-07-15:
When leaving zn_open and zn_close out of the loop. There is no issue. Looks like zn_close is not cleaning up completely.

kydos, 2021-07-16:
Hello @vortex314, thanks for submitting the issue. We'll look into this right off and will keep you posted.

Take Care!

kydos, 2021-07-20:
Hello again @vortex314, I had some time to look at the example you submitted and spotted the issue. In short the ```zn_open``` takes a non const pointer to the config which is an input parameter. This is often the convention used in C to indicate a transfer of ownership -- perhaps we should clearly document this. Thus what happens is that when you do the ```zn_close``` the config memory is released and the next open is going to access freed memory. This can be easily solved by moving the config creation within the loop as shown in the code below.  You may also notice that I've added a ```sleep``` after the ```zn_close```, that is to avoid some warning from the link-state algorithm. 

Finally, as mentioned by @JEnoch, there is no need to create a zenoh session each time you want to publish. Thus in a real-world application you would do open/close outside of your loop.

HTH.

Take Care!

```C
#include <stdio.h>
#include <string.h>
#include <time.h>
#include <unistd.h>
#include "zenoh/net.h"

int main(int argc, char **argv) {
    z_init_logger();

    char *uri = "/demo/example/zenoh-c-pub";
    if (argc > 1) {
        uri = argv[1];
    }
    char *value = "Pub from C!";
    if (argc > 2) {
        value = argv[2];
    }

    while (true) {
        zn_properties_t *config = zn_config_default();
        if (argc > 3) {
            zn_properties_insert(config, ZN_CONFIG_PEER_KEY, z_string_make(argv[3]));
        }
        printf("Openning session...\n");
        zn_session_t *s = zn_open(config);
        if (s == 0) {
            printf("Unable to open session!\n");
            exit(-1);
        }

        printf("Declaring Resource '%s'", uri);
        unsigned long rid = zn_declare_resource(s, zn_rname(uri));
        printf(" => RId %lu\n", rid);
        zn_reskey_t reskey = zn_rid(rid);

        printf("Declaring Publisher on %lu\n", rid);
        zn_publisher_t *pub = zn_declare_publisher(s, reskey);
        if (pub == 0) {
            printf("Unable to declare publisher.\n");
            exit(-1);
        }

        char buf[256];
        for (int idx = 0; idx<2; ++idx) {
            sprintf(buf, "[%4d] %s", idx, value);
            printf("Writing Data ('%lu': '%s')...\n", rid, buf);
            zn_write(s, reskey, (const uint8_t *)buf, strlen(buf));
        }
        zn_undeclare_publisher(pub);
        zn_close(s);
        sleep(1);
    };
    return 0;
}
```

kydos, 2021-07-20:
Guys, I suggest we close this issue an create another on documentation to make sure to clearly document the assumptions on ownership transfer.

Thoughts?

vortex314, 2021-07-20:
ok for me. Thanks for the help.

closed 2021-07-20 by vortex314