eclipse-zenoh/zenoh-c #28: Build failure: "error[E0164]: expected tuple struct or tuple variant, found associated function `nix::Error::Sys`"

JEnoch, 2021-07-15:
zenoh-c "master" branch fails to build with following error:
```bash
error[E0164]: expected tuple struct or tuple variant, found associated function `nix::Error::Sys`
Error:   --> /home/runner/.cargo/registry/src/github.com-1ecc6299db9ec823/shared_memory-0.11.4/src/unix.rs:87:13
   |
87 |         Err(nix::Error::Sys(Errno::EEXIST)) => return Err(ShmemError::MappingIdExists),
   |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `fn` calls are not allowed in patterns
   |
   = help: for more information, visit https://doc.rust-lang.org/book/ch18-00-patterns.html

...
```

JEnoch, 2021-07-15:
This issue comes from a recent change in nix:0.22.0 that breaks the shared-memory:0.11.4 dependency that is used by zenoh. There is already a PR in shared-memory to update it to nix:0.22.0: elast0ny/shared_memory-rs#68.

Waiting this to be sorted out in shared-memory, I temporary pushed the Cargo.lock file, allowing to stick with nix:0.21.0 and make zenoh-c to build successfully.

I let this issue open so we remember to remove the Cargo.lock once the issue is solved in shared-memory.

Mallets, 2022-02-15:
This has been fixed upstream and `Cargo.lock` has been updated since then.