eclipse-zenoh/zenoh-c #29: Improve doc wrt. ownership of parameters

JEnoch, 2021-07-20:
As discussed in https://github.com/eclipse-zenoh/zenoh-c/issues/27#issuecomment-883368889 we should make clear in the API documentation if a function takes the ownership of some parameters (e.g. `zn_open(struct zn_properties_t *config)`).

Also, some functions that take a non-const pointer but not taking ownership of this pointer must be updated to take a const pointer (e.g. `zn_info_as_str(struct zn_session_t *session)`)

Mallets, 2022-04-04:
This issue relates to the issue #53 regarding API alignment. 
The new API being developed should already cover ownership aspects by construction, it just need to be verified that ownership management is clear.

Mallets, 2022-09-12:
Ownership is covered by this RFC: https://github.com/eclipse-zenoh/roadmap/blob/main/rfcs/ALL/Ownership%20in%20C.md

closed 2022-09-12 by Mallets