eclipse-zenoh/zenoh-c #32: add listener properties seems fail

Berniejk, 2021-08-06:
I want to modify the zn_pub.c example to create listener peer with tcp/192.168.50.92:7447 and it seems didn't work but rust sample works. Below is my code added before opening session, did I miss something?
`zn_properties_insert(config, ZN_CONFIG_LISTENER_KEY, z_string_make("tcp/192.168.50.92:7447"));`

Berniejk, 2021-08-07:
> I want to modify the zn_pub.c example to create listener peer with tcp/192.168.50.92:7447 and it seems didn't work but rust sample works. Below is my code added before opening session, did I miss [...]

I found that only windows has such kind of situation, assign listener works at ARM Linux.

Berniejk, 2021-11-03:
I found that it would work if I replace ZN_CONFIG_LISTENER_KEY with 0x42 directly.

p-avital, 2022-04-04:
Hi @Berniejk ,

Sorry this issue fell through, do you still need any help?

Mallets, 2022-09-28:
Forthcoming 0.6.0-beta.1 release introduced a new way to deal with configuration in zenoh-c. 
Closing this issue since it is now outdated.

closed 2022-09-28 by Mallets