eclipse-zenoh/zenoh-c #49: problem with compilation

femust, 2022-03-05:
Hey, I am on the last release branch 0.5 and I am trying to compile and I got the following errors:
```
   Compiling zenoh-c v0.6.0-dev (/home/andrzej/code_workspace/zenoh/zenoh-c)
error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:48:50
   |
48 | pub static Z_CONFIG_MODE_KEY: &c_char = unsafe { &*(b"mode\0".as_ptr() as *const c_char) };
   |                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:51:14
   |
51 |     unsafe { &*(b"connect/endpoints\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:54:14
   |
54 |     unsafe { &*(b"listen/endpoints\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:57:14
   |
57 |     unsafe { &*(b"transport/auth/usrpwd/user\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:60:14
   |
60 |     unsafe { &*(b"transport/auth/usrpwd/password\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:63:14
   |
63 |     unsafe { &*(b"scouting/multicast/enabled\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:66:14
   |
66 |     unsafe { &*(b"scouting/multicast/interface\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:69:14
   |
69 |     unsafe { &*(b"scouting/multicast/address\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:72:14
   |
72 |     unsafe { &*(b"scouting/timeout\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:75:14
   |
75 |     unsafe { &*(b"scouting/delay\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:78:14
   |
78 |     unsafe { &*(b"add_timestamp\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

error[E0658]: dereferencing raw pointers in statics is unstable
  --> src/types.rs:81:14
   |
81 |     unsafe { &*(b"local_routing\0".as_ptr() as *const c_char) };
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: see issue #51911 <https://github.com/rust-lang/rust/issues/51911> for more information

For more information about this error, try `rustc --explain E0658`.
error: could not compile `zenoh-c` due to 12 previous errors
make[2]: *** [CMakeFiles/cargo.dir/build.make:71: CMakeFiles/cargo] Error 101
make[1]: *** [CMakeFiles/Makefile2:94: CMakeFiles/cargo.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

```
do you know what may be the issue?

p-avital, 2022-04-04:
Are you sure you're on a 0.5 branch? 
To be clear, `master` is not a `0.5`, but a preview of `0.6`.

The errors you're getting indicate that you're trying to build 0.6 code, using CMake (introduced in 0.6), with an outdated version of rust. Can you also let us know what version of rust is installed on your machine (use `rustup show` and copy whatever is bellow "active toolchain")?

Mallets, 2022-12-15:
Closing this issue because outdated. Master has already bumped to `v0.7.0-rc`.

closed 2022-12-15 by Mallets