eclipse-zenoh/zenoh-c #59: [Bug] crash at network.rs Line:156

ben56191397, 2022-07-25:
### Describe the bug

I use  zenoh-c (0.5.0-beta.9) ,some times it crash at network.rs:156 .
I want to know,  is zenoh-c session surport multithread ?   or should use  lock when ?use zn_write?

Thanks

### To reproduce

1.start publisher  in multithread
2.start subscriber 
3. test for more time ,about one day

### System info

1.platform :
   Linux mdchost 4.19.0+ #1 SMP Mon Jul 20 14:26:42 CST 2020 aarch64 
    aarch64 aarch64 GNU/Linux
2.cpu info :
    processor	: 0
    BogoMIPS	: 200.00
    Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm jscvt fcma dcpop asimddp asimdfhm
    CPU implementer	: 0x48
    CPU architecture: 8
    CPU variant	: 0x1
    CPU part	: 0xd01
    CPU revision	: 0

3.   zenoh-c branch:
    zenoh-c (0.5.0-beta.9)  :  branch_0.5.0-beta.9 0d15ee4 [origin/branch_0.5.0-beta.9]

crash info:
thread 'async-std/runtime' panicked at "call 'option::unwrap() on a None Value' , /root/XX/github.com-1ecc6299db9ec823/src/net/routing/network.rs:156:59

p-avital, 2022-07-25:
Hi,
zenoh-c does support multithreading, I assigned @OlivierHecart who's the expert on the part of the code where the crash happened :)

ben56191397, 2022-07-25:
thanks ,best wishes

OlivierHecart, 2022-07-26:
Hi @ben56191397 ! 
The problem you experience is related to the linkstate protocol. There has been a lot of changes and bug fixes in the linkstate protocol since version 0.5.0-beta.9. In particular this fix prevents zenoh to crash in case of problems: https://github.com/eclipse-zenoh/zenoh/commit/472cc34acb9d85f781a1c6fe735fa124c88ce97b

This fix is available in the master branch. Note also that we now have a version of zenoh that do not need a linkstate protocol and can perform classic peer-to-peer communication. This version is available in this branch of zenoh: https://github.com/eclipse-zenoh/zenoh/tree/configurable-clique-peers (This branch should be able to compile against zenoh-c api-changes branch: https://github.com/eclipse-zenoh/zenoh-c/tree/api-changes). 
All those branches will be merged into master and released in version 0.6.0-beta.1 that we plan for mid September.

p-avital, 2022-07-26:
Hi @ben56191397,

Just to add a bit to @OlivierHecart's mention of 0.6: `zenoh-c`'s `api-changes` branch is the one which is to become 0.6.0-beta.1 in September, and we're (I think) done with the changes on it as far as API goes. Feel free to check it out and experiment with it if you like to live on the edge. Let us know if you think there are holes in the API that you miss :)

ben56191397, 2022-07-27:
HI @p-avital @OlivierHecart 
    Thank you for your timely reply and problem solving。 We use 0.5 version now,  0.6 is not a final release  version ,if we merge the changes https://github.com/eclipse-zenoh/zenoh/commit/472cc34acb9d85f781a1c6fe735fa124c88ce97b to 0.5,  may it works ?

OlivierHecart, 2022-07-28:
Hi @ben56191397 !
It may indeed work.

ben56191397, 2022-07-29:
Hi @OlivierHecart ,
Thanks, I  try to build it.

ben56191397, 2022-08-26:
It  indeed work, this problem should be closed.  Thanks！

closed 2022-12-15 by Mallets