eclipse-zenoh/zenoh-c #90: [Bug] Compilation error

maxikrie, 2022-11-06:
### Describe the bug

[100%] Running RUSTFLAGS= cargo build --release;--features=logger-autoinit
   Compiling zenoh-c v0.6.0-beta.1 (/home/zenoh-c)
error[E0512]: cannot transmute between types of different sizes, or dependently-sized types
  --> src/get.rs:69:18
   |
69 |         unsafe { std::mem::transmute(val) }
   |                  ^^^^^^^^^^^^^^^^^^^
   |
   = note: source type: `Option<Reply>` (1280 bits)
   = note: target type: `get::z_owned_reply_t` (1472 bits)

For more information about this error, try `rustc --explain E0512`.
error: could not compile `zenoh-c` due to previous error
gmake[2]: *** [CMakeFiles/cargo.dir/build.make:71: CMakeFiles/cargo] Error 101
gmake[1]: *** [CMakeFiles/Makefile2:98: CMakeFiles/cargo.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2

### To reproduce

Follow the install instructions

### System info

rustc 1.65.0 (897e37553 2022-11-02)

maxikrie, 2022-11-06:
I thought that it had to do with the new version of rust.
Awesome, thank you for the quick help!

maxikrie, 2022-11-07:
On branch rustc-1.65, I am getting the following error when trying to compile examples:

zenoh-c/include/zenoh_commons.h:227:3: error: unknown type name ‘u128’
  227 |   u128 _2;
      |   ^~~~

maxikrie, 2022-11-09:
The latest commit on the rustc-1.65 branch (use proper u128 reexport) fixes the issue.

Thanks!

closed 2022-11-09 by maxikrie

p-avital, 2022-11-09:
> The latest commit on the rustc-1.65 branch (use proper u128 reexport) fixes the issue.

Thanks for spotting it in the first place, and sorry for forgetting to notify you after fixing it :)