eclipse-zenoh/zenoh-c #92: [Bug] Compile error under macOS

sanri, 2022-11-10:
### Describe the bug

error info:
```
error[E0512]: cannot transmute between types of different sizes, or dependently-sized types
  --> src/get.rs:69:18
   |
69 |         unsafe { std::mem::transmute(val) }
   |                  ^^^^^^^^^^^^^^^^^^^
   |
   = note: source type:  Option<Reply> (1280 bits)
   = note: target type:  get::z_owned_reply_t (1472 bits)
```

### To reproduce

run cmd 
```
cmake --build /Volumes/Data/projects/zenoh-c/cmake-build-release --target cargo -j 12
```

### System info

- Platform: macOS 12.6.1
- CPU:  x86_64 
- Rust toolchains: rustc 1.66.0-beta.1 (e080cc5a6 2022-11-01)
- zenohc commit:  [cc41a3b5a45330c2e8848e0298c3fc0173e976c4]

p-avital, 2022-11-10:
Hi,
PR #89 fixes this, but I'm having issues with the CI that refuses to use rustc 1.65 (which brought new niche optimizations that caused layout changes that are the root of the error you're getting).
In the meantime, you can check out its corresponding branch `rustc-1.65`

p-avital, 2022-11-10:
#89 Closes this

closed 2022-11-10 by p-avital