eclipse-zenoh/zenoh-c #132: z_undeclare_keyexpr doesn't drop passed keyexpr [Bug]

milyin, 2023-01-16:
### Describe the bug

The function `z_undeclare_keyexpr` accepts mutable reference to `z_owned_keyexpr` and is supposed to delete the passed value. But in fact it clones the passed keyexpr and then drops the clone:
https://github.com/milyin/zenoh-c/blob/master/src/keyexpr.rs#L438-L444

See also discussion in https://github.com/eclipse-zenoh/zenoh-c/pull/131

I see one more problem related to this. The `z_undeclare_keyexpr` accepts only keyexprs of `KeyExprInner::Wire` and `KeyExprInner::BorrowedWire` variants. But there is no way to know is the instance of `z_owned_keyexpr` wire (with `expr_id` or not). I think such function needs to be added to zehoh-c API. 

### To reproduce

-

### System info

-

p-avital, 2023-01-16:
> I see one more problem related to this. The `z_undeclare_keyexpr` accepts only keyexprs of `KeyExprInner::Wire` and `KeyExprInner::BorrowedWire` variants. But there is no way to know is the instance [...]

This would be over-constraining the API to current implementation choices.

The opacity over this is fully wanted to allow future changes, both of the internal representation, and of the behaviours around declaration and undeclaration (which may have additional behaviour that also fits non-wire representation).

p-avital, 2023-01-17:
Closed by #133

closed 2023-01-17 by p-avital