eclipse-zenoh/zenoh-c #135: unify locations of installed header files between zenoh-c and zenoh-pico

milyin, 2023-01-23:
### Describe the bug

The zenoh-c main header is `/usr/local/include/zenohc/zenohc.h`. For zenoh-pico it's `/usr/local/include/zenoh-pico.h`. Better to have the same logic for both projects. Also it's not clear which approach to follow in `zenoh-cpp`

### To reproduce

-

### System info

-

gabrik, 2023-01-25:
This is quite strange, as when installing following the README of zenoh-c, the issue is actually there, while when installing from the package file the include is just `#include <zenoh.>` maybe just the CMake needs to be updated.

milyin, 2023-01-25:
So the correct location of <zenoh.h> is root of "include" directory, correct? And the same logic should be followed in zenon.cpp ? If so, I'll make the PRs for CMakeLists.txt

gabrik, 2023-01-25:
When installing with apt yes, they are actually installed in `/usr/include`. I think the same logic should apply for zenoh-cpp.
@p-avital do you agree?

cguimaraes, 2023-01-25:
I agree on following the same logic.

closed 2023-09-06 by milyin