eclipse-zenoh/zenoh-c #138: add SONAME field to shared library [Bug]

milyin, 2023-02-08:
### Describe the bug

The .so file generated by rust doesn't contain SONAME field. This prevents normal linking to this .so with cmake: it embeds full path to .so file instead of only library name.
This is known not yet fixed issue in Cargo: https://github.com/rust-lang/cargo/issues/5045
Solution:
- use some workaround for now (https://developers.redhat.com/articles/2022/09/05/how-create-c-binding-rust-library#fixing_the_soname)
- keep eye on this issue in Cargo and apply correct solution when it's available 

### To reproduce

_

### System info

Linux platform affected
Apple - probably yes, to be checked
Windows - not affected, but problem with searching DLLs should be resolved too

milyin, 2023-02-08:
Can be fixed also with using `IMPORTED_NO_SONAME` flag in cmake

lu-zero, 2023-02-14:
You may use https://crates.io/crates/cargo-c

milyin, 2023-03-14:
Fixed in https://github.com/eclipse-zenoh/zenoh-c/pull/141

closed 2023-03-14 by milyin

simpleton, 2025-04-21:
Hi @milyin , I follow the build steps mentioned in README.md, but I still produce the so file w/o SONAME on linux.

```
mkdir -p build && cd build 
cmake ../zenoh-c
cmake --build . --config Release
```

do you have any insight on this issue?

```
$ readelf -d ../target/release/libzenohc.so

Dynamic section at offset 0xb55738 contains 30 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [ld-linux-x86-64.so.2]
 0x000000000000000c (INIT)               0xb7000
```

Thank you!