eclipse-zenoh/zenoh-c #140: Transport layer configuration

maxikrie, 2023-02-14:
### Describe the feature

Hello,

From what I understand zenoh operates on works on different network protocols and layers, but I cannot find a documentation on how to configure this (the only documentation I found is https://zenoh.io/blog/2022-08-12-zenoh-serial/). Specifically I would like to choose a UDP/IP connection between a publisher and subscriber. 

Naively I added following code to z_sub.c assuming 'best effort' would choose a UDP connection:

` z_subscriber_options_t opts = z_subscriber_options_default();
opts.reliability = Z_RELIABILITY_BEST_EFFORT;
 z_owned_subscriber_t sub = z_declare_subscriber(z_loan(s), z_keyexpr(expr), z_move(callback), &opts);
`
When running the z_pub/z_sub example, I still see in wireshark that data is exchanged over TCP/IP. Is this expected and/or configuring the protocol not desired?

Thanks for clarification!
Max

cguimaraes, 2023-02-14:
Hi @maxikrie !
If your `z_pub` / `z_sub` are in `client` mode, then you need to run a Zenoh Router (aka `zenohd`) listening on a UDP locator. You can do it either via configuration file or passing it via command line `-l udp/10.0.0.1:7447`. The `z_pub` / `z_sub` will scout for the Zenoh entity to connect to. If you do not scouting procedure to be triggered just statically configure the locator on both of them.

If your `z_pub` / `z_sub` are in `peer` mode, then you need to provide a UDP locator to one or both of them. The rest will happen as described above but without a Zenoh router intermediating the communication.

For a easier and faster support, feel free to join our Discord channel: https://discord.gg/2GJ958VuHs

maxikrie, 2023-02-21:
Hi @cguimaraes !
Thank you for the detailed answer. 

1) Unfortunately  I cannot pass the locator correctly as an argument. I am trying:
`./z_sub demo/example/** udp/192.168.178.32:7447`
returning following error message:

> Couldn't insert value `udp/192.168.178.32:7447` in configuration at `listen/endpoints`. This is likely because `listen/endpoints` expects a JSON-serialized list of strings

2) In the meantime I have tried the above employing python3:
`python3 z_pub.py`  and `python3  z_sub.py -l udp/192.168.178.32:7477`
This works as expected as now I see on wireshark that messages are exchanged via UDP instead of TCP. However, the subscriber has to be started after publisher. Ideally I would like to be able to decide on the subscriber which transport layer to employ even when the publisher is already running.

3) How can I configure the transport layer to be shared memory (when running publisher and subscriber on the same machine) instead of UDP or TCP?

In general, for these type of questions do you prefer me to ask them on discord?

Many thanks!
Max

maxikrie, 2023-02-26:
@cguimaraes I have also posted this question on discord in case you prefer answering there. Thanks!

cguimaraes, 2023-02-27:
1. Something seems to not be aligned with the remaining examples (tagging @p-avital ). For the moment, you can pass the locators as a JSON list of strings:
`z_sub demo/example/** ["tcp/192.168.178.32:7447"]`

2. No matter the order that you start your `pub` and `sub`, they should be able to discover one another and exchange information. Unfortunately, I cannot replicate your issue. If you can provide us more information and logs, we can try to figure it out.

3. If you enabled shared memory features in compilation, then if both `pub` and `sub` are in the same machine they will transparently use shared memory for the communication.

maxikrie, 2023-02-27:
1.  I am on Debian 6.0.12-1~bpo11+1 (2022-12-19) x86_64 GNU/Linux and HEAD detached at 0.7.0-rc. I get 

> ./z_sub demo/example/** ["tcp/192.168.1.2:7447"]
Couldn't insert value `[tcp/192.168.1.2:7447]` in configuration at `listen/endpoints`. This is likely because `listen/endpoints` expects a JSON-serialized list of strings

2. Since I cannot get the above to run, I can only test with the python3 examples. Maybe this is more suited for the zenoh-python repo and I should post this issue there? In any case I get very reproducible results on HEAD detached at 0.7.0-rc
and the latest pip3 install eclipse-zenoh (0.7.0rc0). 

On the same PC I first start python3 z_pub.py and after that python3  z_sub.py -l udp/192.168.1.2:7447. However,  on wireshark I see that messages are passed via TCP. If now close the publisher and start it again, messages are passed as expected via UDP. Attached is a screenshot where I close and restart the publisher after 10s as described.

[image]

cguimaraes, 2023-02-28:
1. Can you try it like this `./z_sub demo/example/** '["tcp/192.168.1.2:7447"]'`?

2. This is because by default, if you do not set the listening endpoints, they will listen on TCP port. So, when you have one entity listening on UDP and another on TCP, the scouting entity will establish a connection with the protocol of the discovered entities. Try setting the listening endpoints also for the pub.

maxikrie, 2023-02-28:
1. That worked!
2. Ok, I think I understand - is the protocol/behavior described anywhere in this detail? I am asking for future reference and better understanding. Also, regarding my use case, naively I would like to have something like a configuration option on the subscriber side where I can decide between reliable (TCP) and best effort (UDP) - disregard the wording if I am mixing up terms - and the specific technicalities like ports are negotiated behind the curtains . What you describe feels like there needs to be a central instance for all the bookkeeping on ports especially when operating on the same machine (e.g. multiple subscribers need to know which ports are already in use).

cguimaraes, 2023-03-01:
`I would like to have something like a configuration option on the subscriber side where I can decide between reliable (TCP) and best effort (UDP)`
I see your point, and I kind of share your view.

To some extent, that is already possible for TCP as the default protocol where new Zenoh peers will use the available ports. The issue as you mention is that when we want to use a different protocol we need to explicit define the full locator to use (which includes the port number).

I propose we move this discussion to Discord. Is it fine for you?

maxikrie, 2023-03-01:
Absolutely! Should I start a new thread or will you do it based on the information I provided? It seems that you understand my request and I am glad you share my view.

cguimaraes, 2023-03-07:
According to @OlivierHecart message in Discord:

> ﻿﻿The default configuration is equivalent to setting this listening endpoint: -l tcp/[::]:0 (or in IPv4 -l tcp/0.0.0.0:0). Port 0 meaning "system chosen port". So to tell Zenoh to use UDP and  and keep the default behavior, it should be sufficient to set the listening port to: `-l udp/[::]:0` (or in IPv4 `-l udp/0.0.0.0:0`).

closed 2023-03-07 by cguimaraes