eclipse-zenoh/zenoh-c #155: z_owned_publisher_t and others stores reference to z_owned_session_t itself [Bug]

milyin, 2023-04-27:
### Describe the bug

When object `z_owned_publisher_t` is created, it stores inside the reference to related `z_owned_session_t` object (which is actually the `Option<Session>`: https://github.com/eclipse-zenoh/zenoh-c/blob/master/src/session.rs#L37)

https://github.com/eclipse-zenoh/zenoh-c/blob/0.7.0-rc/src/publisher.rs#L128
https://github.com/eclipse-zenoh/zenoh/blob/0.7.0-rc/zenoh/src/session.rs#L599

This means that the `z_owned_session_t` can't be moved. The following code crashes:
```C
    z_owned_config_t config = z_config_default();
    z_owned_session_t s = z_open(z_move(config));
    z_owned_publisher_t pub = z_declare_publisher(z_loan(s), z_keyexpr(keyexpr), NULL);

    // Move session to another struct
    z_owned_session_t s2 = s;
    s = z_session_null();

    // Publisher crashes, as it holds a reference to the session
    z_publisher_put_options_t options = z_publisher_put_options_default();
    z_publisher_put(z_loan(pub), (const uint8_t*)"foo", 3, &options);
```

For C++ this leads to impossibility to use move semantics for Session class. 

Session is heavy object which is rarely created/destroyed. And it uses dynamic memory allocation anyway. I think this would be useful to have guarantee that `z_session_object_t` remains valid, no matter where it is moved.

### To reproduce

-

### System info

-

milyin, 2023-04-27:
@p-avital what do you think?

p-avital, 2023-04-27:
The goal was never for the session to be moved around (in fact, its only mover should be its destructor, which should already only run when it isn't aliased), so the fact that the session can't be moved isn't really an issue IMO: it's natural not to be able to move stuff that's aliased.

p-avital, 2023-04-27:
In fact, the problem here isn't moving stuff, it's that a value is getting reassigned while aliased.

milyin, 2023-04-27:
This is exactly how move semantic works in C++: you copy the structure, then clean the original without destruction. So to have move semantic for Session class I should place 'z_owned_session_t` instance to heap and store pointer to it in Session class.

But it's not clear then, why should I do this for Session only? Can I freely move 'z_owned_publisher_t` for example or it also may have some external references to it?

milyin, 2023-04-27:
Here is C++ example which just breaks if you by mistake creates subscriber on local session instead of the one stored in class. And if user forget to make 'CustomerClass' unmovable, he will also get a crash

https://github.com/milyin/zenoh-cpp/blob/move_semantic_fix/examples/simple/universal/z_simple.cxx

This is similar to that @gabrik doing now and this is probably how customers may use our library

p-avital, 2023-04-27:
You're never allowed to move stuff that's aliased at the time of move. That moving the session breaks things that alias it (like a publisher) is normal behaviour.

Pinning the session on the heap is _a_ solution to ensure a stable address, but it's not the only one (it could simply be assigned once to a static and never moved, for example).

To me, there's nothing special to do, it's just that the Session is especially prone to being aliased. If one wants it pinned (and possibly co-owned), they should allocate for it themselves (so that we don't presume which way they'd want to manage that allocation)

p-avital, 2023-04-27:
To be clear, it's perfectly valid to move a Session around, as long as it isn't referenced by something else during the move, just like it's perfectly valid to move an array around until one of its cells is pointed at by something else. This isn't anything new, it's just how aliasing works.

To do otherwise, we'd need to smart-pointer all the things... Arguably, we allow this with `Arc<Session>` in Rust having extended behaviour... Maybe we should have a way to expose that `Arc<Session>` through `zenoh-c`, but then what about pico? :/

milyin, 2023-04-27:
How the user of the library knows, is the specific owned object aliased or not? We don't have lifetimes in C and C++. 
So we either can enable move semantic for C++ wrappers of zenoh-c structures and say to customer "hey, you can move owned structures, unless you called this, this and this methods on them, otherwise all will crash at any moment". 
Or we can completely disable move semantic for owned structures and say: if you want to do anything with them, use std::unique_ptr.

milyin, 2023-04-27:
When this aliacing is actually happens? Can anything except Sesssion be aliased?

p-avital, 2023-04-27:
I'd say we should document better when loans are long-term aliases (i.e. a reference is held past the function call).

Disabling moves completely is an option, but it seems to radical to me.

And yes, anything can be aliased, since for all we know the user might just decide to hold a pointer to any given thing at any given time. We just need to be more explicit about when we do.

milyin, 2023-04-27:
If user holding the pointer, it's his responsibility. But if he calls method 'declare_publicher`, how can he know that aliasing happens at this moment?
So I think that at least for C++ every class which can create aliasing to itself should store aliased data on the heap. Or forbid move semantic, pinning itself to memory.

milyin, 2023-04-27:
But it the problem is actually is only related to `z_owned_session_t` and nothing else, why not to make zenoh-c less restrictive and decide that no structure can store direct reference to another structure (but it's ok to store refrerence to heap data of another structure) ? 
How significantly such rule may affect zenoh-c - what do you think?

milyin, 2023-04-27:
I think there are two problems mixed. I don't mean `use-after-remove` problem here. This can be solved by reference counting or by accurate coding, but it's not related to original problem.

The original problem is this: structure, which 
- contains word `owned` in it's name
- can be created only by special constructor
- should be dropped by special 'drop' function 
imposes additional restriction: it also can't be moved from it's place in memory

This restriction is very unobvious. Developer knows, that the structure (Session) have some state, stored outside of the structure (otherwise special constructor/destructor would not be necessary). So developer normally expects, that when some other structure based on this structure is created (Publisher), it may hold reference to this outside-of-structure stored state. But he definitely do not expect, that the derived structure will hold reference to the stack-allocated owned structure itself. This just doesn't make sense - what's the reason to have some external state and do not use it?

kydos, 2023-05-02:
> The goal was never for the session to be moved around (in fact, its only mover should be its destructor, which should already only run when it isn't aliased), so the fact that the session can't be moved isn't really an issue IMO: it's natural not to be able to move stuff that's aliased.

I am not sure that restriction is acceptable, nor I think that the inability of moving a session was a requirement, since this creates mismatch even with Rust. I think we need to address this issue.

milyin, 2023-06-08:
fixed in https://github.com/eclipse-zenoh/zenoh-c/pull/157

closed 2023-06-08 by milyin