eclipse-zenoh/zenoh-c #164: [Bug] Windows debug build tries to link release DLL

junglie85, 2023-06-01:
### Describe the bug

Building a shared debug DLL on Windows tries to load `zenohc.dll` instead of `zenohcd.dll`.

### To reproduce

```
cmake ..\zenoh-c\ -GNinja -DCMAKE_BUILD_TYPE=Debug
cmake --build . --config Debug
cmake --build . --config Debug --target install
```

Link error that unable to find `zenohc.dll`. No issues if I do a release build.

### System info

Windows 10. 
zenoh-c commit: `57d69f0a0ef89d59308be578b5e59c8f06fd8fa8`

junglie85, 2023-06-01:
Here's the output of `dumpbin /HEADERS C:\Dev\zenoh-debug\lib\zenohcd.dll.lib` showing this:

```
Microsoft (R) COFF/PE Dumper Version 14.35.32217.1
Copyright (C) Microsoft Corporation.  All rights reserved.

Dump of file C:\Dev\zenoh-debug\lib\zenohcd.dll.lib

File Type: LIBRARY

FILE HEADER VALUES
            8664 machine (x64)
               3 number of sections
        EEB4EBFF time date stamp
             10A file pointer to symbol table
               8 number of symbols
               0 size of optional header
               0 characteristics

SECTION HEADER #1
.debug$S name
       0 physical address
       0 virtual address
      40 size of raw data
      8C file pointer to raw data (0000008C to 000000CB)
       0 file pointer to relocation table
       0 file pointer to line numbers
       0 number of relocations
       0 number of line numbers
42100040 flags
         Initialized Data
         Discardable
         1 byte align
         Read Only

SECTION HEADER #2
.idata$2 name
       0 physical address
       0 virtual address
      14 size of raw data
      CC file pointer to raw data (000000CC to 000000DF)
      E0 file pointer to relocation table
       0 file pointer to line numbers
       3 number of relocations
       0 number of line numbers
C0300040 flags
         Initialized Data
         4 byte align
         Read Write

SECTION HEADER #3
.idata$6 name
       0 physical address
       0 virtual address
       C size of raw data
      FE file pointer to raw data (000000FE to 00000109)
      E0 file pointer to relocation table
       0 file pointer to line numbers
       0 number of relocations
       0 number of line numbers
C0200040 flags
         Initialized Data
         2 byte align
         Read Write

FILE HEADER VALUES
            8664 machine (x64)
               2 number of sections
        CBBD6A7E time date stamp
              B8 file pointer to symbol table
[... 2032 lines cut ...]
  Type         : code
  Name type    : name
  Hint         : 176
  Name         : zc_shmbuf_ptr

  Version      : 0
  Machine      : 8664 (x64)
  TimeDateStamp: C4CB30CF
  SizeOfData   : 00000020
  DLL name     : zenohc.dll
  Symbol name  : zc_shmbuf_set_length
  Type         : code
  Name type    : name
  Hint         : 177
  Name         : zc_shmbuf_set_length

  Summary

          C0 .debug$S
          14 .idata$2
          14 .idata$3
           8 .idata$4
           8 .idata$5
           C .idata$6
```

milyin, 2023-09-06:
@junglie85 Thank you for your report! I've reproduced the issue. Yes, it's a bug: `zenoh.dll` and `zenoh.dll.lib` are renamed to `zenohd.dll` and `zenohd.dll.lib` for installation of debug variant, but `zenohd.dll.lib` still points to `zenoh.dll`
This can be fix it either by making Rust build into `zenohd.dll` or patching `zenohd.dll.lib` after installation. Looking for a better way to do it now.

milyin, 2023-09-08:
Fixed by updating lib name Cargo.toml depending on debug/release configurations

closed 2023-09-08 by milyin