eclipse-zenoh/zenoh-c #165: [Bug] Access violation on Windows builds

junglie85, 2023-06-01:
### Describe the bug

Running a simple publisher on Windows fails with an access violation accessing location 0xFF whereas the exact same program on WSL (Ubuntu) runs as expected.

### To reproduce

Start a subscriber:

In WSL, build the `zenoh-c` library as per the repo instructions then compile the following program:
```
#include <stdio.h>
#include <string.h>

#include "zenoh.h"

int main(int argc, char **argv)
{
    printf("Running...\n");

    char *keyexpr = "demo/example/zenoh";
    char *value = "Hello Zenoh!";

    printf("A\n");
    z_owned_config_t config = z_config_default();
    if (zc_config_insert_json(z_loan(config), Z_CONFIG_CONNECT_KEY, "[\"tcp/localhost:7447\"]") < 0)
    {
        printf("Could not add endpoint to config");
        exit(-1);
    }
    printf("B\n");

    z_owned_session_t session = z_open(z_move(config));
    if (!z_check(session))
    {
        printf("Unable to open session");
        exit(-1);
    }

    z_owned_publisher_t publisher = z_declare_publisher(z_loan(session), z_keyexpr(keyexpr), nullptr);
    if (!z_check(publisher))
    {
        printf("Unable to declare publisher");
        exit(-1);
    }

    z_publisher_put_options_t options = z_publisher_put_options_default();
    z_publisher_put(z_loan(publisher), (const uint8_t *)value, strlen(value), &options);

    z_undeclare_publisher(z_move(publisher));
    z_close(z_move(session));

    printf("Done\n");

    return 0;
}
```
Build and run:
```
gcc main.cpp -L$(pwd) -lzenohc -o zc
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$(pwd) ./zc
```
See output and message in subscriber:
```
Running...
A
B
Done
```

Repeat, but in Windows:
```
cl.exe main.cpp /o zc.exe /link zenohc.dll.lib
zc.exe
Running...
A

```

### System info

- Windows 10 (native and WSL Ubuntu 22.04).
- zenoh-c 57d69f0a0ef89d59308be578b5e59c8f06fd8fa8

junglie85, 2023-06-01:
I've discovered that if I replace `Z_CONFIG_CONNECT_KEY` with the expected value `"connect/endpoints"` it does not crash.

milyin, 2023-09-10:
@junglie85, thank you for the report! Reproduced the issue in https://github.com/eclipse-zenoh/zenoh-c/tree/pub_crash_on_windows branch (z_crash example), investigating

milyin, 2023-09-11:
Problem doesn't happen if examples are built with static zenoh library:
```
cmake -Szenoh-c -Bbuild -DZENOHC_BUILD_EXAMPLES_WITH_STATIC_LIB=TRUE
```
But for dynamic library case the problem exists:
static linking:
```
Z_CONFIG_CONNECT_KEY=00007FF787FE7381
*Z_CONFIG_CONNECT_KEY=connect/endpoints
```
dynamic linking:
```
Z_CONFIG_CONNECT_KEY=CCCC0000CA5F25FF
(crash)
```
Seems like the way how it is exported from rust is incorrect:
```Rust
#[no_mangle]
pub static Z_CONFIG_CONNECT_KEY: &c_char =
    unsafe { &*(b"connect/endpoints\0".as_ptr() as *const c_char) };
```

milyin, 2023-09-12:
Problem occurs on all constants, not only on string ones. This code:
```C
    printf("Z_ROUTER = %d\n", Z_ROUTER);
    printf("Z_PEER = %d\n", Z_PEER);
    printf("Z_CLIENT = %d\n", Z_CLIENT);
```
also outputs wrong values:
```
Z_ROUTER = -829610497
Z_PEER = -835312129
Z_CLIENT = -835443201
```

The working solution is to add `__declspec(dllimport)` to declarations. After this change:
```C
__declspec(dllimport) extern const unsigned int Z_ROUTER;
__declspec(dllimport) extern const unsigned int Z_PEER;
extern const unsigned int Z_CLIENT;
```
the output is
```
Z_ROUTER = 1
Z_PEER = 2
Z_CLIENT = -854252033
```

Here are some information which may help:
https://github.com/rust-lang/rust/issues/27438
https://www.reddit.com/r/rust/comments/169yeei/linkage_to_dll/
https://rust-lang.github.io/rfcs/2627-raw-dylib-kind.html

It may be also make sense to consider `raw-dylib` option for windows.

closed 2023-09-19 by milyin