eclipse-zenoh/zenoh-c #178: [Bug] Build error for previously released versions after new protocol merge

Yadunund, 2023-09-12:
### Describe the bug

Attempting to build `zenoh-c` from source for any version/tag other than `master` produces the build error below. My project was previously using `0.7.2-rc` and I assumed the [new protocol](https://github.com/eclipse-zenoh/zenoh/pull/183) changes would only affect `master/main`? Is it the intention to force users to switch to the new protocol?

Or is it an oversight in the [Cargo.toml.in](https://github.com/eclipse-zenoh/zenoh-c/blob/590bea934f7642261405ae8cbfa8bf4834d0bfe5/Cargo.toml.in#L50-L52) where the "branch" should be `"@CARGO_PROJECT_VERSION@"` instead of `master`? That way older versions could create bindings for the older version of the zenoh protocol.

<details>
<summary>Click for detailed build error</summary>

```
   Compiling zenoh-c v0.10.0-dev (/Users/yadunund/zenoh-c/build)
error[E0512]: cannot transmute between types of different sizes, or dependently-sized types
   --> /Users/yadunund/zenoh-c/zenoh-c/src/commons.rs:113:30
    |
113 |             _owner: unsafe { std::mem::transmute(buf) },
    |                              ^^^^^^^^^^^^^^^^^^^
    |
    = note: source type: `zenoh::zenoh_buffers::ZBuf` (320 bits)
    = note: target type: `[usize; 4]` (256 bits)

error[E0616]: field `buf` of struct `ZSlice` is private
   --> /Users/yadunund/zenoh-c/zenoh-c/src/commons.rs:132:65
    |
132 |             let start_offset = unsafe { start.offset_from(slice.buf.as_slice().as_ptr()) };
    |                                                                 ^^^ private field

error[E0616]: field `buf` of struct `ZSlice` is private
   --> /Users/yadunund/zenoh-c/zenoh-c/src/commons.rs:134:47
    |
134 |             *slice = match ZSlice::make(slice.buf.clone(), start_offset, start_offset + len) {
    |                                               ^^^ private field

error[E0512]: cannot transmute between types of different sizes, or dependently-sized types
   --> /Users/yadunund/zenoh-c/zenoh-c/src/commons.rs:225:26
    |
225 |         _owner: unsafe { std::mem::transmute(buf) },
    |                          ^^^^^^^^^^^^^^^^^^^
    |
    = note: source type: `zenoh::zenoh_buffers::ZBuf` (320 bits)
    = note: target type: `[usize; 4]` (256 bits)

error[E0308]: mismatched types
   --> /Users/yadunund/zenoh-c/zenoh-c/src/scouting.rs:115:17
    |
114 |             _pid: match h.zid {
    |                         ----- this expression has type `ZenohId`
115 |                 Some(id) => unsafe { std::mem::transmute(id) },
    |                 ^^^^^^^^ expected `ZenohId`, found `Option<_>`
    |
    = note: expected struct `ZenohId`
                 found enum `Option<_>`

error[E0308]: mismatched types
   --> /Users/yadunund/zenoh-c/zenoh-c/src/scouting.rs:116:17
    |
114 |             _pid: match h.zid {
    |                         ----- this expression has type `ZenohId`
115 |                 Some(id) => unsafe { std::mem::transmute(id) },
116 |                 None => z_id_t { id: [0; 16] },
    |                 ^^^^ expected `ZenohId`, found `Option<_>`
    |
    = note: expected struct `ZenohId`
                 found enum `Option<_>`

error[E0277]: the trait bound `WhatAmIMatcher: From<u32>` is not satisfied
   --> /Users/yadunund/zenoh-c/zenoh-c/src/scouting.rs:252:41
    |
252 |     let what = WhatAmIMatcher::try_from(config.zc_what).unwrap_or(WhatAmI::Router | WhatAmI::Peer);
    |                ------------------------ ^^^^^^^^^^^^^^ the trait `From<u32>` is not implemented for `WhatAmIMatcher`
    |                |
    |                required by a bound introduced by this call
    |
    = help: the trait `From<WhatAmI>` is implemented for `WhatAmIMatcher`
    = note: required for `u32` to implement `Into<WhatAmIMatcher>`
    = note: required for `WhatAmIMatcher` to implement `TryFrom<u32>`

error[E0277]: the trait bound `WhatAmIMatcher: From<u32>` is not satisfied
   --> /Users/yadunund/zenoh-c/zenoh-c/src/scouting.rs:252:16
    |
252 |     let what = WhatAmIMatcher::try_from(config.zc_what).unwrap_or(WhatAmI::Router | WhatAmI::Peer);
    |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `From<u32>` is not implemented for `WhatAmIMatcher`
    |
    = help: the trait `From<WhatAmI>` is implemented for `WhatAmIMatcher`
    = note: required for `u32` to implement `Into<WhatAmIMatcher>`
    = note: required for `WhatAmIMatcher` to implement `TryFrom<u32>`

Some errors have detailed explanations: E0277, E0308, E0512, E0616.
For more information about an error, try `rustc --explain E0277`.
error: could not compile `zenoh-c` (lib) due to 8 previous errors
make[2]: *** [target/release/libzenohc.dylib] Error 101
make[1]: *** [CMakeFiles/cargo.dir/all] Error 2
make: *** [all] Error 2
```
</details>

### To reproduce

Checkout `zenoh-c` to `0.10.0-dev` or `0.7.2-rc` and build as per instructions in README.

### System info

- Ubuntu 22.04 / MacOS Ventura
- rustc v1.72

milyin, 2023-09-12:
Created tag https://github.com/eclipse-zenoh/zenoh-c/releases/tag/0.7.2-rc.1, branch https://github.com/eclipse-zenoh/zenoh-c/tree/release-0.7.2-rc.1 with fix. @Mallets., @OlivierHecart , please check

Mallets, 2023-09-12:
@milyin can you create a PR so it's easier to comment/track?

milyin, 2023-09-13:
Done - https://github.com/eclipse-zenoh/zenoh-c/pull/179
PR is just to ease compare: as far as I understand the update can be kept in this [release-0.7.2-rc.1](https://github.com/eclipse-zenoh/zenoh-c/tree/release-0.7.2-rc.1) branch

Mallets, 2023-09-13:
@p-avital can you have a look?

milyin, 2023-09-13:
@p-avital confirmed the fix.
I'm merged the branch into https://github.com/eclipse-zenoh/zenoh-c/tree/release-0.7.2-rc. Anyway the important thing is not the branch itself, but the https://github.com/eclipse-zenoh/zenoh-c/releases/tag/0.7.2-rc.1 tag
I think we need some explicit rules how we manage tags/branches for old releases. Should we keep one branch for each release forever or should we keep tags only and create ad-hoc branches from these tags when we need to make patches to old releases?

milyin, 2023-10-24:
Fixed in tag https://github.com/eclipse-zenoh/zenoh-c/releases/tag/0.7.2-rc.1

closed 2023-10-24 by milyin