eclipse-zenoh/zenoh-c #186: [Bug] Connection between to apps takes long time with interface is active but not transmitting

MelamudMichael, 2023-10-29:
### Describe the bug

Connection between to apps takes long time with interface is active but not transmitting 

### To reproduce

example of an interface:

br-d74a19a0be09: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default
    link/ether 02:42:8b:c0:39:65 brd ff:ff:ff:ff:ff:ff
    inet 172.18.0.1/16 brd 172.18.255.255 scope global br-d74a19a0be09
       valid_lft forever preferred_lft forever
    inet6 2001:db8::1/112 scope global tentative
       valid_lft forever preferred_lft forever
    inet6 fe80::1/64 scope link tentative
       valid_lft forever preferred_lft forever

### System info

running redhat 9 docker image

MelamudMichael, 2023-10-29:
@OlivierHecart  FYI , i opened this one for tracking

Mallets, 2023-11-20:
Fixed by https://github.com/eclipse-zenoh/zenoh/pull/574. Feel free to reopen in case it is not working as expected.

closed 2023-11-20 by Mallets