eclipse-zenoh/zenoh-c #202: [Bug] No error/log print when using shared memory and shared memory is not configured

MelamudMichael, 2023-11-26:
### Describe the bug

When using shared memory over zenoh - additional configuration needs to be made, if not made the data will be sent over socket , and this will result with poor performance.

The expectation is either print a warning of wrong usage or fail the shared memory API

### To reproduce

N/A

### System info

N/A

cvilas, 2023-11-27:
I posted [a note](https://discord.com/channels/914168414178779197/1177925026729709568) on Discord about this and was advised to post it here. So here goes:

The [C example](https://github.com/eclipse-zenoh/zenoh-c/blob/54cbaf5a5118788e4e93b14756d0e153e65861c0/examples/z_pub_shm.c#L37) and [C++ example](https://github.com/eclipse-zenoh/zenoh-cpp/blob/main/examples/zenohc/z_pub_shm.cxx) for publisher with shared memory diverge in the sense that the C-example modifies the config to enable shared memory while the C++ example doesn't. Both repos lack a corresponding subscriber example. In developing my own examples of [publisher here](https://github.com/cvilas/grape/blob/ipc/modules/common/ipc/examples/zenoh/zenoh_pub_shm.cpp) and [subscriber here](https://github.com/cvilas/grape/blob/ipc/modules/common/ipc/examples/zenoh/zenoh_sub.cpp), I noticed that to make shared-memory work, I had to:

1. Modify the config on both publisher and subscriber to enable the shared memory transport
2. On the publisher side, write into shared memory buffers created by the ShmManager (Changing config alone is not enough)
3. On the subscriber side, nothing special was required other than modified config

I hope my understanding is correct. Monitoring the communication on wireshark, it seems to do the right thing, which is:

a) If the publisher and subscriber are on the same host, TCP is still used, but to notify subscriber of which shm segment to use. The subscriber then reads the data off the shm segment directly.
b) If publisher and subscriber are on different hosts, the subscriber still receives the message over TCP
c) If there are two subscribers, one on the same host as publisher and another subscriber on a different host, in wireshark I see two messages from the publisher - one containing the shm segment id (for the subscriber on the localhost) and one containing the full message (for the subscriber on the remote host)

I like the fact that if shared memory is not available between the pub and sub, the system automatically falls back to transport over the network layer.

milyin, 2023-12-01:
This was already done for zenoh-c, as comment above says. But for zenoh-cpp this should be done too

closed 2023-12-01 by milyin