eclipse-zenoh/zenoh-c #211: z_encoding_prefix_t is not correctly transmitted in for queries

ulricheck, 2023-12-03:
### Describe the bug

If a client issues a query and set's the encoding_prefix for the payload value, the receiving queryable receives a different value for encoding_prefix that what was set in the query.

### To reproduce

Run the following example that is compose of zenoh-c examples (z_query + z_get).
```
#include <iostream>

#include <stdio.h>
#include <string.h>
#if defined(WIN32) || defined(_WIN32) || defined(__WIN32) && !defined(__CYGWIN__)
#include <windows.h>
#define sleep(x) Sleep(x * 1000)
#else
#include <unistd.h>
#endif
#include "zenoh.h"

const char *expr = "demo/example/zenoh-c-queryable";
const char *value = R"json({"some":"json", "struct":"payload"})json";
z_keyexpr_t keyexpr;

void query_handler(const z_query_t *query, void *context) {
    z_value_t payload_value = z_query_value(query);
    auto* output = static_cast<z_encoding_prefix_t *>(context);
    if (output != NULL) {
        *output = payload_value.encoding.prefix;
    }
    z_query_reply_options_t options = z_query_reply_options_default();
    options.encoding = z_encoding(Z_ENCODING_PREFIX_TEXT_PLAIN, NULL);
    z_query_reply(query, z_keyexpr(expr), (const unsigned char *)value, strlen(value), &options);
}

int main() {
    z_owned_config_t config = z_config_default();

    printf("Opening session...\n");
    z_owned_session_t s = z_open(z_move(config));
    if (!z_check(s)) {
        printf("Unable to open session!\n");
        exit(-1);
    }
    keyexpr = z_keyexpr(expr);
    if (!z_check(keyexpr)) {
        printf("%s is not a valid key expression", expr);
        exit(-1);
    }

    printf("Declaring Queryable on '%s'...\n", expr);
    z_encoding_prefix_t query_payload_encoding_prefix;
    z_owned_closure_query_t callback = z_closure(query_handler, NULL, &query_payload_encoding_prefix);
    z_owned_queryable_t qable = z_declare_queryable(z_loan(s), keyexpr, z_move(callback), NULL);
    if (!z_check(qable)) {
        printf("Unable to create queryable.\n");
        exit(-1);
    }

    printf("Sending Query '%s'...\n", expr);
    z_owned_reply_channel_t channel = zc_reply_fifo_new(16);
    z_get_options_t opts = z_get_options_default();
    if (value != NULL) {
        opts.value.payload = (z_bytes_t){.len = strlen(value), .start = (uint8_t *)value};
        opts.value.encoding = z_encoding(Z_ENCODING_PREFIX_TEXT_JSON, "");
    }
    z_get(z_loan(s), keyexpr, "", z_move(channel.send),
          &opts);  // here, the send is moved and will be dropped by zenoh when adequate
    z_owned_reply_t reply = z_reply_null();
    for (z_call(channel.recv, &reply); z_check(reply); z_call(channel.recv, &reply)) {
        if (z_reply_is_ok(&reply)) {
            z_sample_t sample = z_reply_ok(&reply);
            z_owned_str_t keystr = z_keyexpr_to_string(sample.keyexpr);
            printf(">> Received ('%s': '%.*s')\n", z_loan(keystr), (int)sample.payload.len, sample.payload.start);
            z_drop(z_move(keystr));
        } else {
            printf("Received an error\n");
        }
    }
    z_drop(z_move(reply));
    z_drop(z_move(channel));

    if (query_payload_encoding_prefix != Z_ENCODING_PREFIX_TEXT_JSON) {
        printf("Query received incorrect z_encoding_prefix_t\n");
    }

    z_undeclare_queryable(z_move(qable));
    z_close(z_move(s));

    return 0;
}
```

On my machine I see the following output:

```
Opening session...
Declaring Queryable on 'demo/example/zenoh-c-queryable'...
Sending Query 'demo/example/zenoh-c-queryable'...
>> Received ('demo/example/zenoh-c-queryable': '{"some":"json", "struct":"payload"}')
Query received incorrect z_encoding_prefix_t
```

### System info

Host: macOS Sonoma 14.1
C++ Standard: 17
Compiler: Apple clang version 15.0.0 (clang-1500.0.40.1)
Target: arm64-apple-darwin23.1.0
Zenoh Version: zenoh-c 0.10.0-rc

p-avital, 2023-12-04:
Hi Ulrich,

Thanks for letting us know, the encoding was indeed not passed on, PR #212 will fix that :)

ulricheck, 2023-12-04:
thanks for the super quick response/fix!

closed 2023-12-04 by p-avital