eclipse-zenoh/zenoh-c #233: Deploying zenoh-C to conan-central index

mmichaelGM, 2024-01-17:
### Describe the feature

Deploying zenoh-C to conan-central index

oteffahi, 2024-04-10:
The situation regarding Conan Center Index is the following:
- Zenoh-C recipe that builds from source requires the Rust compiler, but Rust Toolchain has not yet been integrated on CCI. A PR for a Rust recipe was opened by the CCI community on November 8th 2023, it remains open to this day as maintainers seem to be hesitant in supporting Rust on CCI. Regardless, a Zenoh-C recipe PR that builds from source was opened by a CCI community member, and it depends on the Rust recipe.
- We have looked into making a Zenoh-C recipe that uses our pre-compiled release artefacts from the Zenoh-C repo. For the GNU Linux target, the library is compiled on Ubuntu 20 which dynamically links it with `glibc 2.29`. The issue is that CCI runners are on Ubuntu 18 and Ubuntu 16, which are respectively on `glibc 2.27` and `glibc 2.25` ([more detail here](https://github.com/conan-io/conan-docker-tools/tree/001c761387deee2a00864ed12630bcb053727126/modern#library-versions)): this effectively means that their CI will never successfully build and test our recipe even if we explicitely test for glibc version and gracefully exit the recipe if the version does not match. We are still exploring the possibility of building and releasing artefacts linked on older ubuntu versions (@fuzzypixelz can go into more details regarding this)

@MelamudMichael Could you please elaborate on your use-cases for Zenoh-C's Conan Recipe? Is CCI the only Conan registery that would fit these use-cases?

MelamudMichael, 2024-04-11:
I think that it can be on any other Conan registry, we just need an official Conan recepie that we can use for cross compiling and etc