eclipse-zenoh/zenoh-c #242: iOS/Android support

jakobturftank, 2024-02-06:
### Describe the feature

We are a robotics company that have considering migrating our communication channels to Zenoh, especially after it was announced as a new rmw for ROS2. 

However, we have decided against it, after we found that we could not build the C bindings for iOS and Android. We are using flutter and dart so we typically use C libraries with dart ffi.

I hope that we could have the C bindings buildable from Android and iOS sometime in the future.

Thank you.

Mallets, 2024-02-06:
Hello @jakobturftank , 

Would you mind describing more in details what didn't work out in your attempt to build the C bindings for iOS and Android?
It would be great if you could provide some guidance on how to setup the environment and reproduce the issue.

andreasturftank, 2024-02-06:
Hello,
As we are using Flutter for some UI, we would like to interface with Zenoh from Dart. We imagine the easiest way to do so, is to use the c-interopability of Dart to call the functions of zenoh-c from a dynamic library compiled for the specific platform (iOS/Android). The actual calling of the dynamic library occurs via bindings (Dart:FFI) which are generated from the zenoh-c header files.

To this end, I have been following [this tutorial](https://docs.flutter.dev/platform-integration/android/c-interop) and by extension [this one](https://developer.android.com/studio/projects/add-native-code) to build an android library which contains both the C library and the corresponding Dart bindings.

I have run into issues when building from the CMakeLists from Gradle. Specificially, I run into an `ld` error, claiming that the `.../app/.cxx/Debug/694h3a66/arm64-v8a/debug/target/debug/libzenohcd.so is incompatible with aarch64linux`, but I suspect that something else is wrong. I can see from [this roadmap discussion](https://github.com/eclipse-zenoh/roadmap/discussions/98#discussioncomment-7568542) that you haven't tried to build zenoh-c for Android possibly due to Android NDK issues - did you ever get further than this?

I have not attempted to build for iOS, yet. Do you have this use case on your roadmap or a iOS friendly Zenoh implementation such as in Swift or Objective-C?

Mallets, 2024-02-07:
Chiming in @milyin and @p-avital

uupks, 2024-02-19:
With my attempt to cross-compile zenoh-c to the Android system, the biggest problem is that Android system does not support POSIX shared memory.  But shared-memory enabled for zenoh even if zenoh-c "shared-memory" feature is disabled. This is to make "std::mem::transmute" work for `ZSLice`..

uupks, 2024-02-19:
When I try to cross-compile zenoh-c to aarch64-linux-android, the following error will be reported: 
```
   Compiling shared_memory v0.12.4
error: shared_memory isnt implemented for this platform...
  --> /home/r22052/.cargo/registry/src/index.crates.io-6f17d22bba15001f/shared_memory-0.12.4/src/lib.rs:43:9
   |
43 |         compile_error!("shared_memory isnt implemented for this platform...");
   |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0433]: failed to resolve: use of undeclared crate or module `os_impl`
   --> /home/r22052/.cargo/registry/src/index.crates.io-6f17d22bba15001f/shared_memory-0.12.4/src/lib.rs:121:27
    |
121 |                     match os_impl::create_mapping(&cur_id, self.size) {
    |                           ^^^^^^^ use of undeclared crate or module `os_impl`

error[E0433]: failed to resolve: use of undeclared crate or module `os_impl`
   --> /home/r22052/.cargo/registry/src/index.crates.io-6f17d22bba15001f/shared_memory-0.12.4/src/lib.rs:130:38
    |
130 |             Some(ref specific_id) => os_impl::create_mapping(specific_id, self.size)?,
    |                                      ^^^^^^^ use of undeclared crate or module `os_impl`

error[E0433]: failed to resolve: use of undeclared crate or module `os_impl`
   --> /home/r22052/.cargo/registry/src/index.crates.io-6f17d22bba15001f/shared_memory-0.12.4/src/lib.rs:207:19
    |
207 |             match os_impl::open_mapping(unique_id, self.size, &self.ext) {
    |                   ^^^^^^^ use of undeclared crate or module `os_impl`

error[E0433]: failed to resolve: use of undeclared crate or module `os_impl`
  --> /home/r22052/.cargo/registry/src/index.crates.io-6f17d22bba15001f/shared_memory-0.12.4/src/lib.rs:55:10
   |
55 |     ext: os_impl::ShmemConfExt,
   |          ^^^^^^^ use of undeclared crate or module `os_impl`

error[E0433]: failed to resolve: use of undeclared crate or module `os_impl`
   --> /home/r22052/.cargo/registry/src/index.crates.io-6f17d22bba15001f/shared_memory-0.12.4/src/lib.rs:232:14
    |
232 |     mapping: os_impl::MapData,
    |              ^^^^^^^ use of undeclared crate or module `os_impl`

For more information about this error, try `rustc --explain E0433`.
error: could not compile `shared_memory` (lib) due to 6 previous errors
warning: build failed, waiting for other jobs to finish...
```

uupks, 2024-06-07:
With this https://github.com/eclipse-zenoh/zenoh-c/commit/390595b246dc6bbb51043d2dcecd79eba262fc55  commit, we can cross compile for `aarch64-linux-android`  successfully

closed 2025-12-16 by sashacmc