eclipse-zenoh/zenoh-c #245: shared memory is not freed after termination[Bug]

MelamudMichael, 2024-02-20:
### Describe the bug

shared memory is not freed after termination

l /dev/shm/
total 2596
drwxrwxrwx  2 root   root       6560 Feb 19 16:57 ./
drwxr-xr-x 16 root   root       4340 Feb 13 09:27 ../
-rw-------  1 xz62px unix_users 4104 Feb 19 15:57 shmem_106608B83E24F29E
-rw-------  1 xz62px unix_users 4104 Feb 14 09:08 shmem_109653567179D5DF
-rw-------  1 xz62px unix_users 4104 Feb 18 07:49 shmem_11F2668C7808FC3A
-rw-------  1 xz62px unix_users 4104 Feb 18 09:06 shmem_138F914CE12C2A2C
-rw-------  1 xz62px unix_users 4104 Feb 14 11:13 shmem_13C383C9E7BC17CE
-rw-------  1 xz62px unix_users 4104 Feb 14 08:58 shmem_146B1378BBE5398A

### To reproduce

see above

### System info

any

yellowhatter, 2024-02-28:
Hello, @MelamudMichael !
Can you please describe your testing scenario to achieve this?

yellowhatter, 2024-03-04:
Hello, @MelamudMichael !

In actual release filesystem-mapped shm segments are really not erased in case of owner process termination. The new SHM codebase in the upcoming release will have this fixed.

DenisBiryukov91, 2025-04-07:
Fixed in the current release

closed 2025-04-07 by DenisBiryukov91