eclipse-zenoh/zenoh-c #294: [Bug] When using unixpipe transport, killing an application stops remote applications.

OlivierHecart, 2024-03-21:
### Describe the bug

When using unixpipe transport, killing an application stops remote applications.

### To reproduce

1. Build examples with unixpipe feature.
  `mkdir build` 
  `cd build` 
  `cmake .. -DZENOHC_CARGO_FLAGS="--features=zenoh/transport_unixpipe"` 
  `cmake --build . --target examples`
2. Start `z_sub` with unixpipe listening endpoint.
  `./target/release/examples/z_sub "demo/example/**" "[\"unixpipe/test.pipe\"]"`
3. Start `z_pub` with unixpipe connect endpoint.
  `./target/release/examples/z_pub "demo/example/zenoh-c-pub" "Pub from C" 0 "[\"unixpipe/test.pipe\"]"`
4. Kill `z_pub`
5. `z_sub` exits.

### System info

- Platform: MacOS
- CPU: M2
- zenoh-c version: commit `bd0fe98772b7d65e8fb71eb0369d7c85c9b09c73`

yellowhatter, 2024-03-21:
I think we need to investigate the pipe spec on it's behavior when other pipe end closes. That might be the reason of this bug.

yellowhatter, 2025-04-07:
According to @DenisBiryukov91 , this is still reproduced. To be validated

yellowhatter, 2025-04-07:
Reproduced on zenoh-c but no reproduction on zenoh-rust

yellowhatter, 2025-04-07:
zenoh-rust example can cause zenoh-c example to exit as described, but not vice versa.

yellowhatter, 2025-04-07:
Okay, got the issue.
The problem is in SIGPIPE signal that is sent by default to process when it tries to write to closed unixpipe.
In rust application, SIGPIPE is implicitly ignored: https://github.com/rust-lang/rust/issues/62569
...that's why there was no issues when testing unixpipe with Rust examples.

However, our C application is not "rusty" and probably doesn't execute Rust's `_start` that is intended to set SIGPIPE ignored.

I think I need to investigate what is missing and make sure Rust's `_start` functionality is executed for zenoh-c

yellowhatter, 2025-04-07:
Oookay, the objdump shows that zenoh-c binary lacks `lang_start` function that is responsible for doing Rust stuff before `main()`.

I will investigate on how can we add it to our C binaries as this will be the most proper way to address the original issue

yellowhatter, 2025-04-07:
So, to be short, it seems that we need to manually call `std::rt::init()` somewhere before using our C bindings.

PR #984 (2025-04-08): Rust entrypoint
Fixes #294 

Adds entrypoint `alternative_rusty_entry_point()` and installs SIGPIPE to be ignored there

Tested manually with static and shared library
closed 2025-04-08 by yellowhatter