eclipse-zenoh/zenoh-c #317: [Bug] Compile error under WSL

cngzhnp, 2024-04-09:
### Describe the bug

Try to compile with CMake 3.28.1

Error info:

```
error[E0512]: cannot transmute between types of different sizes, or dependently-sized types
  --> /home/cngzhnp/.cpm/zenoh-c/b977a0d3736aca1b432d213551a801f1f22d34c4/src/lib.rs:80:26
   |
80 |                 unsafe { std::mem::transmute::<$src_type, $dst_type>(self) }
   |                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
  ::: /home/cngzhnp/.cpm/zenoh-c/b977a0d3736aca1b432d213551a801f1f22d34c4/src/get.rs:63:1
   |
63 | impl_guarded_transmute!(ReplyInner, z_owned_reply_t);
   | ---------------------------------------------------- in this macro invocation
   |
   = note: source type: `Option<zenoh::query::Reply>` (1920 bits)
   = note: target type: `get::z_owned_reply_t` (1792 bits)
   = note: this error originates in the macro `impl_guarded_transmute` (in Nightly builds, run with -Z macro-backtrace for more info)
   ```

### To reproduce

```
cmake .. -DZENOHC_BUILD_IN_SOURCE_TREE=ON -DZENOHC_LIB_STATIC=ON
make -j12
```

### System info

- Platform : Ubuntu 23.10
- WSL : 6.1.21.2-microsoft-standard-WSL2+
- CMake : 3.28.1
- Rust Toolchain : stable-x86_64-unknown-linux-gnu - rustc 1.77.1 (7cf61ebde 2024-03-27)
- Zenohc : 0.10.1-rc

milyin, 2024-04-09:
Hello @cngzhnp,
Would you mind to try to use rust toolchain 1.72 ?I hope with it the compilation should pass.
The root cause of the issue is that that for maximizing speed in zenoh-c it was decided to work rust structures directly, without any c-repr wrappers. This allows to avoid extra indirection and memory allocations. But the cost is that we have to have pair of structures: rust one which is actually handled by library (`Option<zenoh::query::Reply>`) and C one, which is allocated by C compiler (`z_owned_reply_t`). These structures have the same size and alignment and converted back and forth by `std::mem::transmute`. With new Rust compiler version the layout of rust structure may change and this causes compilation failure.
Supposedly at some moment we'll implement automatic generation of C structure by rust one, but it's not easy, considering that this should work for cross compilation too.

cngzhnp, 2024-04-09:
Hello @milyin,

Even though, when I changed to 1.72.0 toolchain, build does not allow me to end up successfully. Here is the output of `rustup show` prints out below:

```Default host: x86_64-unknown-linux-gnu
rustup home:  /home/cngzhnp/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu
1.72.0-x86_64-unknown-linux-gnu (default)

active toolchain
----------------

1.72.0-x86_64-unknown-linux-gnu (default)
rustc 1.72.0 (5680fa18f 2023-08-23)
```

Still I got the same error on my machine.

cngzhnp, 2024-04-09:
I think that problem is solved with help of this commit but it does not belong to **0.10.1-rc** tag. So, when I set back zenoh-c project to main branch, it works.

[Rust 1.77 toolchain - Fix alignment and layout for z_owned_reply_t](https://github.com/eclipse-zenoh/zenoh-c/pull/295)

milyin, 2024-04-09:
@cngzhnp thank you for the solution! That's still strange for me why 0.10.1-rc fails with rust 1.72, I'll try to investigate if I find the time

doganulus, 2024-04-12:
I managed to build `zenohc` using the following `Dockerfile`. I think the problem was related to `ZENOHC_CARGO_CHANNEL`, which was default to `stable` but it should be `1.72`. I think a recent commit fixed this issue. It may be better to setup CI for recent released versions. Especially versions matching those in Zenoh containers. This is why I wanted to stick to `0.10.1-rc`.

```dockerfile
FROM buildpack-deps:bookworm

ARG ZENOH_VERSION=0.10.1-rc
ARG ZENOHC_VERSION=${ZENOH_VERSION}

ARG RUST_VERSION=1.72.1
ARG RUST_PROFILE=minimal
ARG RUST_TOOLCHAIN=${RUST_VERSION}-x86_64-unknown-linux-gnu
ENV CARGO_HOME="/usr/local/cargo"
ENV RUSTUP_HOME="/usr/local/rustup"
ENV PATH "/usr/local/cargo/bin:/usr/local/rustup:${PATH}"

RUN curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | \
    sh -s -- -y --no-modify-path --profile ${RUST_PROFILE} --default-toolchain ${RUST_VERSION}

RUN git clone --depth 1 --branch ${ZENOHC_VERSION} https://github.com/eclipse-zenoh/zenoh-c.git /tmp/zenoh-c && \
    cmake -S/tmp/zenoh-c -B/tmp/zenoh-c/build \
        -DCMAKE_BUILD_TYPE=Release \
        -DZENOHC_CARGO_CHANNEL=${RUST_TOOLCHAIN} \
        -DZENOHC_INSTALL_STATIC_LIBRARY=TRUE \
        && \
    cmake --build /tmp/zenoh-c/build/ --target install -j$(nproc) && \
    rm -rf /tmp/zenoh-c
```

DenisBiryukov91, 2024-07-16:
The issue should no longer arise in dev/1.0.0 branch since the sizes and alignments of zenoh-rust types are no longer hardcoded, but fetched from rust compiler.

closed 2024-11-14 by milyin