eclipse-zenoh/zenoh-c #352: z_open should accept z_open_options_t to allow for future extensions

DenisBiryukov91, 2024-05-06:
### Describe the release item

z_open signature should be changed to 
`z_error_t z_open(struct z_owned_session_t *this_, struct z_owned_config_t *config, const z_open_options_t *options);`

This will allow providing more options in the future without breaking the api.

milyin, 2024-05-29:
There was also a request from @kydos to provide `open` function without required `config` parameter, as usually our config is default: https://github.com/eclipse-zenoh/zenoh/issues/805, This is not decided yet and may be it makes sense to think about these issues together.

On the zenoh-c side we can put pointer to config inside options structure, allow it to be nullptr and ,provide `z_open_options_default` with null pointer to config. Or better name it `z_session_options_t` ?

The code will look like:
```C
z_owned_session_t session;
int err = z_open(&session, z_open_options_default());
```

Here I assume that we passing options by value, which is not the case at this moment. Issue #401 was created for this

milyin, 2024-05-30:
```C
z_owned_session_t session;
z_open_options_t opts;
z_open_options_default(&ots);
int err = z_open(&session, &opts);
```

DenisBiryukov91, 2024-05-30:
To allow providing options inline and specifying only non-default values we can probably try to use macros as proposed here https://stackoverflow.com/questions/13716913/default-value-for-struct-member-in-c.

milyin, 2024-09-03:
All session open parameters now comes through config, seems it's controversial idea to complexify z_open with additional parameter. If we need this in future, we can make addition open function with "options"

milyin, 2024-09-10:
accordingly to @Mallets comment in discord: same thing for `z_close. It should accept `z_close_options_t` which is empty for now.

closed 2024-09-12 by milyin