eclipse-zenoh/zenoh-c #374: add methods to `z_id_t`

milyin, 2024-05-26:
### Describe the release item

- `into_keyexpr`: http://github.com/eclipse-zenoh/zenoh/blob/c-coverage/commons/zenoh-protocol/src/core/mod.rs#L86
- `MAX_SIZE`, `to_le_bytes`: http://github.com/eclipse-zenoh/zenoh/blob/c-coverage/commons/zenoh-protocol/src/core/mod.rs#L78
- `rand` http://github.com/eclipse-zenoh/zenoh/blob/c-coverage/commons/zenoh-protocol/src/core/mod.rs#L82
- `size` http://github.com/eclipse-zenoh/zenoh/blob/c-coverage/commons/zenoh-protocol/src/core/mod.rs#L73

DenisBiryukov91, 2024-05-29:
I'm not sure they make much sense. 
Rand is done probably for testing purposes only, size is always 16, to_le_bytes is actually what c/cpp user already gets.
Not sure what would be the use for into_keyexpr.

milyin, 2024-05-29:
@DenisBiryukov91 agree. These methods were already decided some time ago to be removed from public API. See issue https://github.com/eclipse-zenoh/zenoh/issues/806

closed 2024-05-29 by milyin