eclipse-zenoh/zenoh-c #403: Zenoh unresolved reference to shm in Android 13.

Mallets, 2024-05-30:
### Discussed in https://github.com/eclipse-zenoh/roadmap/discussions/138

<div type='discussions-op-text'>

<sup>Originally posted by **ashirvad-acsia** May 29, 2024</sup>
While adopting the zenoh pub sub middleware implementation for Android 13 we are facing the following issues while integrating zenoh library:
Undefined reference  to shm - shared memory APIs :shared_memory::unix::create_mapping::he2aa66c0237b54c1: error: undefined reference to 'shm_open'
To resolve this issue we tried to disable the shared memory feature in the cmakelist file by the following modification:
declare_cache_var(ZENOHC_BUILD_WITH_SHARED_MEMORY FALSE BOOL "Enable shared-memory zenoh-c feature")  
However the build error remains same for the shm dependency.
Could anybody please suggest a solution?

We are using the following environment:
Zeno C API with a C++ wrapper for our middleware C++ implementation
Target : Android 13 - running in an emulator</div>

Mallets, 2024-05-30:
@yellowhatter this issue seems to be blocked by the capability of zenoh-c to fully disable shared-memory support.

yellowhatter, 2024-05-30:
@milyin @DenisBiryukov91 FYI

sjames, 2024-06-02:
> @yellowhatter this issue seems to be blocked by the capability of zenoh-c to fully disable shared-memory support.

Being able to build this natively in Android is pretty important.  Thanks for moving it to the roadmap discussions for 1.0.0.

yellowhatter, 2024-06-02:
@Mallets @sjames 
WIP: https://github.com/eclipse-zenoh/zenoh-c/issues/406

Mallets, 2024-09-13:
Closed by #406

closed 2024-09-13 by Mallets