eclipse-zenoh/zenoh-c #438: Memory leak after undeclaring publisher, subscriber and closing session with c bindings

hguelle, 2024-06-11:
### Describe the bug

I have some google tests which also checks the memory leaks. Tests open basically a session and create a publisher and a subscriber. After the test they should be cleaned up again. For that, I am using :

- int8_t z_undeclare_publisher(struct z_owned_publisher_t *publisher);
- int8_t z_undeclare_subscriber(struct z_owned_subscriber_t *sub);
- int8_t z_close(struct z_owned_session_t *session);

These are all returning 0 as ret val. But after the test 

### To reproduce

1. Start a session
2. declare publisher
3. declare subscriber
4. undeclare publisher
5. undeclare subscriber
6. close session
7. memory leak detected 

[memory_leaks.txt](https://github.com/user-attachments/files/15784298/memory_leaks.txt)

### System info

Debian x86_64 GNU/Linux

DenisBiryukov91, 2024-06-12:
These memory "leaks" come from rust static variables (namely ZRuntimePool) which are never dropped. 
Rust does not drop static variables by design.
We have an API to force-drop ZRuntimePool to prevent corresponding memory leaks from appearing in the logs, but it is only available in zenoh-rust. We will likely provide similar api for zenoh-c/zenoh-c++ for the next release.

DenisBiryukov91, 2024-08-09:
@hguelle If possible, could you check the dev/1.0.0 version and call  zc_stop_z_runtime() at the end of your program and let us know if you still get memory leaks reported.

PR #584 (2024-08-09): add valgrind memory leaks check
add valgrind memory leaks checks;
fixed memory leak in handlers drop function;
add zc_stop_z_runtime() function, that can be used to free zenoh-related static variables;
Should resolve #438 
Closes #582.
closed 2024-10-14 by DenisBiryukov91