eclipse-zenoh/zenoh-c #486: remove constructor of `z_entty_global_id_t` and it's fields access

milyin, 2024-06-28:
### Describe the release item

In the existing Rust API the `EntityGlobalId` is a wrapper fo `EntityGlobalIdProto`. All details of `EntityGlobalIdProto` is hidden now. So functions `z_entity_global_id_new`, `z_entity_global_id_zid`, `z_entity_global_id_eid` should also be removed from C API

milyin, 2024-06-28:
@Mallets please confirm, that now `EntityGlobalId` is blind and user should not neither be able to create it manually nor have access to (SessionId, EntityId) fields
@wyfo, than you for noticing this issue in zenoh-c API

Mallets, 2024-06-28:
`EntityGlobalId` is internal and users should only have read access to it. In any case, `EntityGlobalId` should only be present in `SourcInfo` which is `unstable` if I'm not mistaken.

milyin, 2024-06-28:
Should be fixed in https://github.com/eclipse-zenoh/zenoh-c/pull/485

milyin, 2024-08-04:
Rust's `EntityGlobalId` provides RO access to zid and eid fields. Constructor is removed. So C API is ok,

closed 2024-08-04 by milyin