eclipse-zenoh/zenoh-c #528: Support `zc_reply_fifo_new_unbounded`

YuanYuYuan, 2024-07-17:
### Describe the feature

A handler to an unbounded flume channel was supported in 0.11.0 but missing in 1.0.0.

DenisBiryukov91, 2024-07-17:
In 0.11.0 unbounded channel was created by passing a 0 channel size, now passing 0 creates a rendezvous channel. I  believe Zenoh rust also does not support unbounded flume-channel as its predefined handler. It was discussed with @Mallets and agreed that it does not worth implementing unbounded channel in zenoh-c.

YuanYuYuan, 2024-07-17:
I see. Then let's close this issue 😁.

wyfo, 2024-07-17:
Unbounded is rarely a good idea. A lot of channels uses the same implementation for both bounded and unbounded, so there is no difference in memory use. `flume` is just a mutex around a `VecDeque` (I'm quite disappointed to discover that), so keeping it bounded to avoid overflowing the memory is always better, and passing `usize::MAX` has pretty much the same effect than as an unbounded one.