eclipse-zenoh/zenoh-c #618: [Bug] consider rust-toolchain.yaml when building from other directory

milyin, 2024-08-23:
### Describe the bug

See fix https://github.com/eclipse-zenoh/zenoh-c/pull/617
This fix makes ad-hoc update which fixes specific issue. 
@yellowhatter commented that that the real cause of the problem is probably ignoring rust-toolchain.yaml when building from other directory. Need to be revised

### To reproduce

-

### System info

-

yellowhatter, 2024-08-23:
I faced the same error on RMW and fixed it occasionally by making 1.75 toolchain default. This is a sign that rust-toolchain.yaml is definitely ignored.

The official doc is not clear about this, but people say that Cargo respects rust-toolchain.yaml only in CWD

yellowhatter, 2024-08-23:
@milyin please talk to Martin Geoff ( @gmartin82 ) about this: https://github.com/ZettaScaleLabs/zenoh-c/commit/b84e333ef6f88ef2cee0d94afd36da279456dae0

gmartin82, 2024-08-23:
Also see [this PR](https://github.com/eclipse-zenoh/zenoh-c/pull/616) which copies the toolchain config file into the build directory to ensure cargo picks it up.

milyin, 2024-09-02:
Validated with https://crates.io/crates/version_check library that requested compiler version is used

closed 2024-09-02 by milyin