eclipse-zenoh/zenoh-c #646: make `z_string_array_t` methods unstable

milyin, 2024-09-04:
### Describe the release item

The `z_owned_string_array_t` is used in single place: as destination type for `z_hello_locators` function. It's never accepted by any zenoh function and therefore there is no reason for user to be able to create and modify it.
So functions `z_string_array_new`, `z_string_array_push_by_alias`, `z_string_array_push_by_copy` seems to be not really needed. Zenoh-pico doesn't provide these functions.

The proposal is to make these methods unstable in zenoh-c to not force us to implement them in zenoh-pico.

DenisBiryukov91, 2024-09-04:
we could technically use z_owned_string_array_t instead of *const *const c_char in z_config_client when passing array of peers - this would allow accepting non-null-terminated strings (this might also be interesting for c++ to allow accepting a list of string view instead of vector of std::string in the corresponding function - but performance gain would likely be minimal)

milyin, 2024-09-05:
> we could technically use z_owned_string_array_t instead of *const *const c_char + length in z_config_client when passing array of peers - this would allow accepting non-null-terminated strings (this might also be interesting for c++ to allow accepting a list of string view instead of vector of std::string in the corresponding function - but performance gain would likely be minimal)

@sashacmc what do you think? This makes sense for me: the `const char *const *peers, size_t n_peers` is really ugly, I like the idea to use `z_owned_sting_array_t` here instead