eclipse-zenoh/zenoh-c #740: [Bug] Failed build zenoh-c with unstable api for arm-unknown-linux-gnueabi

muzili, 2024-10-04:
### Describe the bug

As the title said, failed to build zenoh-c library with `ZENOHC_BUILD_WITH_UNSTABLE_API` flag, there's no this issue if  `ZENOHC_BUILD_WITH_UNSTABLE_API` is set to false.

### To reproduce

1. `export RUSTFLAGS="-Clinker=/usr/bin/arm-linux-gnueabi-gcc -Car=/usr/bin//usr/bin/arm-linux-gnueabi-ar"`
2. `mkdir build && cd build && cmake -DZENOHC_BUILD_WITH_UNSTABLE_API=true -DZENOHC_BUILD_WITH_SHARED_MEMORY=true -DZENOHC_CUSTOM_TARGET=arm-unknown-linux-gnueabi .. && make -j16`
```
error[E0080]: evaluation of constant value failed
   --> src/shm/provider/types.rs:195:1
    |
195 | / decl_c_type!(
196 | |     owned(z_owned_chunk_alloc_result_t, option ChunkAllocResult),
197 | |     loaned(z_loaned_chunk_alloc_result_t),
198 | | );
    | |_^ the evaluated program panicked at 'Size mismatch: type z_loaned_chunk_alloc_result_t has size 16 while type z_owned_chunk_alloc_result_t has size 20', src/shm/provider/types.rs:195:1
    |
    = note: this error originates in the macro `$crate::panic::panic_2021` which comes from the expansion of the macro `decl_c_type` (in Nightly builds, run with -Z macro-backtrace for more info)
```

### System info

- Platform: Debian 12
- Version: zenoh-c 1.0.0.10

yellowhatter, 2024-10-18:
it looks like ZENOHC_CUSTOM_TARGET is not passed to opaque-types

yellowhatter, 2024-10-23:
@muzili please check, now everything should be fixed in main

closed 2024-10-23 by yellowhatter

muzili, 2024-10-24:
> @muzili please check, now everything should be fixed in main

Verified, thanks. :+1: