eclipse-zenoh/zenoh-c #785: make rust analyzer's "go to definition" on C loaned, owned, etc. types go to corresponding `decl_c_type!(...)`

milyin, 2024-10-22:
### Describe the feature

Working on zenoh-c it's frequently necessary to know how rust and c-wrapper types are related.
For example: 
- I see type `z_owned_bytes_t` in the code
- I want to know how what this type means
- I press "F12" and see this blind type in opaque-types module
  ```rust
  #[repr(C, align(8))]
  pub struct z_owned_bytes_t {
    _0: [u8; 40],
   }
   ```
- but what I really need is this:
   ```rust
    decl_c_type! {
       owned(z_owned_bytes_t, ZBytes),
       loaned(z_loaned_bytes_t),
   }
   ```

This may be disappointing to anyone who is trying to understand how zenoh-c works. This will slowdown our own performance, making learning curve for zenoh-c internals too steep for new developers.

The proposal is to declare some internal types in opaque_types instead of the final ones. And declare real API types `z_owned_bytes_t`, `z_loaned_bytes_t`, etc. inside `decl_c_type!` macro. In this case "go to definition" will go directly to this macro call with all necessary information

DenisBiryukov91, 2024-10-24:
I think the behavior as is is correct and does not need any change, if needed you can always search for something like "owned(z_owned_bytes_t".

milyin, 2024-10-24:
> I think the behavior as is is correct and does not need any change, if needed you can always search for something like "owned(z_owned_bytes_t".

It's correct but not convenient. When I go to type definition, I want to know how to use this type. The opaque type is completely blind and doesn't give any hints where to find additional information. So this is additional obstacle for ones who want to understand how the API works