eclipse-zenoh/zenoh-c #798: Hybrid Communication

owny990312, 2024-10-31:
### Describe the feature

1.Is it possible to use zenoh for both shared memory communication and network cross-machine communication?Does it currently have such basic functionality?
2.Can it even autonomously schedule which solution to use?
Thanks

YuanYuYuan, 2024-10-31:
Hi @owny990312! Yes, Zenoh has already supported this functionality. Assuming you have three zenoh sessions assigned as below, you can check the messages are successfully transmitted. 

```mermaid
flowchart TB
    subgraph Machine1["Machine 1"]
        A["Publisher"]
        B["Subscriber 1"]
        A -- SHM --> B
    end
    subgraph Machine2["Machine 2"]
        C["Subscriber 2"]
    end
    A -- TCP --> C
```

- Machine 1
```console
./z_pub_shm
./z_sub_shm
```
- Machine 2
```console
./z_sub_shm
```

owny990312, 2024-10-31:
Thank you for your answer.
 I'd like to further confirm: I just use the shm-API for all cases, then it will automatically handle network communication for cross-machine scenarios while using shared memory for local machine communication. 

Could you point me to any relevant documentation?

owny990312, 2024-10-31:
> Hi @owny990312! Yes, Zenoh has already supported this functionality. Assuming you have three zenoh sessions assigned as below, you can check the messages are successfully transmitted.嗨！是的，Zenoh 已经支持这项功能。假设你有三个 zenoh 会话，如下所示，你可以检查消息是否成功传输。
> 
> ```mermaid
> flowchart TB
>     subgraph Machine1["Machine 1"]
>         A["Publisher"]
>         B["Subscriber 1"]
>         A -- SHM --> B
>     end
>     subgraph Machine2["Machine 2"]
>         C["Subscriber 2"]
>     end
>     A -- TCP --> C
> ```
> * Machine 1 机器 1
> 
> ```
> ./z_pub_shm
> ./z_sub_shm
> ```
> 
> * Machine 2 机器 2
> 
> ```
> ./z_sub_shm
> ```

Thank you for your answer.

I'd like to further confirm: I just use the shm-API for all cases, then it will automatically handle network communication for cross-machine scenarios while using shared memory for local machine communication.

Could you point me to any relevant documentation?

owny990312, 2024-10-31:
> Hi @owny990312! Yes, Zenoh has already supported this functionality. Assuming you have three zenoh sessions assigned as below, you can check the messages are successfully transmitted.嗨！是的，Zenoh [...]

I did some testing and the results were very exciting.

 I'd like to know how the internal judgment is made. 

Also, during execution, since I'm using the shm-API but actually receiving cross-machine data, is this error a normal phenomenon? Will it affect the speed?

owny990312, 2024-10-31:
> Hi @owny990312! Yes, Zenoh has already supported this functionality. Assuming you have three zenoh sessions assigned as below, you can check the messages are successfully transmitted.嗨！是的，Zenoh [...]
I did some testing and the results were very exciting.

I'd like to know how the internal judgment is made.

Also, during execution, since I'm using the shm-API but actually receiving cross-machine data, is this error a normal phenomenon? Will it affect the speed?

[image]

YuanYuYuan, 2024-11-12:
Hi @owny990312, Thanks for reporting this. I've created an issue for it, #814. Don't worry; this doesn't affect performance.

The mechanism of zenoh deciding whether to send the buffer with SHM depends on the underlying link while it's established. You can check the details here https://github.com/eclipse-zenoh/zenoh/blob/e73a89d3bab5d3960159ddfc0ef315407d2fdf7e/io/zenoh-transport/src/shm.rs?plain=1#L123