eclipse-zenoh/zenoh-c #809: SHM API Example

owny990312, 2024-11-08:
### Describe the release item

[image]
In this picture, we can see, it use the following code,

 const z_loaned_shm_t *shm = NULL;
        if (z_bytes_to_loaned_shm(z_sample_payload(sample), &shm) == Z_OK) {
            payload_type = "SHM";
        }

, which can transport the payload into shm, but the "shm" never used in future. Does this example can show subscribe data by shm?

YuanYuYuan, 2024-11-12:
Hi @owny990312, it's a good question. We use the `z_bytes_to_loaned_shm` to check the SHM availability. If the conversion is successful, the variable `shm` will borrow the value from the payload. In the meantime, we still can access the payload. On the other hand, there is `z_bytes_to_owned_shm` that will take the ownership.

YuanYuYuan, 2024-11-13:
@yellowhatter shall we rename this function to `z_bytes_as_loaned_shm` to prevent any possible misleading? 

- `z_bytes_as_loaend_shm` means we borrow it as a loaned shm
- `z_bytes_to_owned_shm` means we convert it to a owned shm

owny990312, 2024-11-13:
> Hi @owny990312, it's a good question. We use the `z_bytes_to_loaned_shm` to check the SHM availability. If the conversion is successful, the variable `shm` will borrow the value from the payload. In the meantime, we still can access the payload. On the other hand, there is `z_bytes_to_owned_shm` that will take the ownership.Hi ，这是一个好问题。我们使用 `z_bytes_to_loaned_shm` 来检查 SHM 的可用性。如果转换成功，变量 `shm` 将从 payload 中借用值。同时，我们仍然可以访问 payload 。另一方面，有 `z_bytes_to_owned_shm` 会获得所有权。

Thanks for your reply. When I use the SHM API of Zenoh C, I find that the documentation is limited. Regarding the following APIs:
- z_shm_provider_garbage_collect()
- z_shm_provider_defragment()
- z_shm_provider_alloc_gc_defrag_dealloc()
I would like to know if these APIs release specific memory blocks or all unused memory. Is there any issue with my understanding?

owny990312, 2024-11-13:
> > @yellowhatter shall we rename this function to `z_bytes_as_loaned_shm` to prevent any possible misleading?我们是否应该将这个函数重命名为`z_bytes_as_loaned_shm`以防止任何可能的误导？
> > 
> > * `z_bytes_as_loaend_shm` means we borrow it as a loaned shmz_bytes_as_loaend_shm 意味着我们将其作为借来的 shm 使用
> > * `z_bytes_to_owned_shm` means we convert it to a owned shmz_bytes_to_owned_shm 意味着我们将其转换为一个拥有的 shm
> 
> Agree, I will rename this同意，我将会重命名这个

Thanks for your reply. I want to ask, if I don't use the API z_bytes_as_loaned_shm, will the sample( subscriber get) obtain data from shared memory or the network?

yellowhatter, 2024-11-13:
> Thanks for your reply. I want to ask, if I don't use the API z_bytes_as_loaned_shm, will the sample( subscriber get) obtain data from shared memory or the network?

Once you get the sample - it is already "transported" to your context. No matter how you are going to use ZBytes, if it has underlying shm data, the shm optimization is already applied.

yellowhatter, 2024-11-13:
> By the way, I have a question: what data do the three APIs z_shm_provider_garbage_collect(), z_shm_provider_defragment(), and z_shm_provider_alloc_gc_defrag_dealloc() release? I'm actually quite confused. 

Well, it's a problem of lacking docs. I will update it
- `z_shm_provider_garbage_collect()` collects all dereferenced SHM buffers and reclaims their memory. The same action can be taken during allocation when necessary in *_gc_* allocation functions.
 
- `z_shm_provider_defragment()` triggers memory defragmentation in the provider backend's allocator. The real actions taken depend on backend's allocator implementation. For current default backend (posix shm provider backend) it will do some job to join adjacent free chunks. As with GC, the same action can be taken during allocation when necessary in *_defragment_* allocation functions. These API just provides you with some fine-grained control on allocator's operation costs and allows to pay these costs in specific moment other then allocation.

- `z_shm_provider_alloc_gc_defrag_dealloc()` this is an allocation API that performs garbage collection and\or defragmentation and\or forced deallocation if needed. Forced deallocation is a tricky thing to deallocate buffer that is referenced and in-use and requires some things to be done at subscriber side do be safe. Do not use it until you need it.

> When I publish data and the subscriber subscribes to it, how is the original data released? If multiple threads subscribe to a topic concurrently, how should resources be released to avoid conflicts and overwriting?
Buffers are reference-counted and normal release happens during garbage collection.

owny990312, 2024-11-13:
> Well, it's a problem of lacking docs. I will update it
>
> * `z_shm_provider_garbage_collect()` collects all dereferenced SHM buffers and reclaims their memory. The same action can be taken during [...]

thanks for your reply.if I just publish data but not subscriber data on shm ，how can the shm release. Only subscribe the data then let reference-counter decrease to zero， the shm can be released？

PR #816 (2024-11-14): to_loaned_shm -> as_loaned_shm 
rename `to_loaned_shm` -> `as_loaned_shm`  to prevent any possible misleading

[closes #809]
closed 2024-11-14 by yellowhatter