eclipse-zenoh/zenoh-c #812: [Bug] The toolchain assumption is too aggressive

YuanYuYuan, 2024-11-11:
### Describe the bug

We assume the Linux platform which compiles zenoh-c for `target.x86_64-unknown-linux-gnu` has _x86_64-linux-gnu-gcc_ installed. This is true for Ubuntu but not for Fedora and NixOS. 

Take the snapshot of Fedora as an example,

```console
[root@8ecbb0d88312 /]# ls -l /usr/bin/x86_64-*
-rwxr-xr-x 4 root root 1213816 Sep 12 00:00 /usr/bin/x86_64-redhat-linux-c++
-rwxr-xr-x 4 root root 1213816 Sep 12 00:00 /usr/bin/x86_64-redhat-linux-g++
-rwxr-xr-x 3 root root 1209704 Sep 12 00:00 /usr/bin/x86_64-redhat-linux-gcc
-rwxr-xr-x 3 root root 1209704 Sep 12 00:00 /usr/bin/x86_64-redhat-linux-gcc-14
-rwxr-xr-x 1 root root     615 Aug  7 00:00 /usr/bin/x86_64-redhat-linux-gnu-pkg-config
```

The prefix is **x86_64-redhat-linux** rather than **x86_64-linux**. On the other hand, NixOS doesn't even add a prefix for gcc.

### To reproduce

skip

### System info

skip

YuanYuYuan, 2024-11-13:
@DenisBiryukov91 do you mind having a look at this issue? Thanks

DenisBiryukov91, 2024-11-13:
@YuanYuYuan Do you mean the assumption in .cargo/config.toml or in cmake toolchain files ?

YuanYuYuan, 2024-11-13:
> @YuanYuYuan Do you mean the assumption in .cargo/config.toml or in cmake toolchain files ?

I mean the file _.cargo/config.toml_. On the other hand, it's fine to set it in _ci/toolchains/TC-x86_64-unknown-linux-gnu.cmake_ since it runs on a Ubuntu runner.

PR #815 (2024-11-14): do not hardcode linker path in .cargo/config.toml
do not hardcode linker path in .cargo/config.toml.
Resolves #812.
closed 2024-11-14 by milyin