eclipse-zenoh/zenoh-c #814: [Bug] The SHM buffer check error log is misleading

YuanYuYuan, 2024-11-12:
### Describe the bug

When sending SHM buffers across machines, we check the SHM buffer availability on the receiver side with [this function](https://github.com/eclipse-zenoh/zenoh-c/blob/153ad8dc5b17d874ded973cc62dddbe56eefa746/src/zbytes.rs?plain=1#L120). However, this reports an error that is [NOT shown in zenoh rust example](https://github.com/eclipse-zenoh/zenoh/blob/e73a89d3bab5d3960159ddfc0ef315407d2fdf7e/examples/examples/z_sub_shm.rs?plain=1#L105). Since we had already properly returned the result in this function, do we need to duplicate this error in the log? 

### To reproduce

1. Build the examples _z_pub_shm_ and _z_sub_shm_.
2. Run the two examples on different hosts.
3. The error log is shown below.
```console
2024-11-12T06:35:36.001040Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.0.109:34133
2024-11-12T06:35:36.001490Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: zenohd listening scout messages on 224.0.0.224:7446
Declaring Subscriber on 'demo/example/**'...
Enter 'q' to quit...
2024-11-12T06:35:36.176089Z ERROR rx-1 ThreadId(07) zenohcd::zbytes: Failed to convert the payload
>> [Subscriber] Received PUT ('demo/example/zenoh-rs-pub': '[  91] Pub from Rust SHM!') [RAW]
2024-11-12T06:35:37.175526Z ERROR rx-1 ThreadId(07) zenohcd::zbytes: Failed to convert the payload
>> [Subscriber] Received PUT ('demo/example/zenoh-rs-pub': '[  92] Pub from Rust SHM!') [RAW]
```

### System info

- zenoh-c: 153ad8dc5b17d874ded973cc62dddbe56eefa746

yellowhatter, 2024-11-13:
@YuanYuYuan I thing we are using this: https://github.com/eclipse-zenoh/zenoh-c/blob/153ad8dc5b17d874ded973cc62dddbe56eefa746/src/zbytes.rs?plain=1#L188

And yes, IMHO we should:

1. Rename all "as_shm*" to "try_as_shm*" in API
2. Do not post error messages for "try_as_shm"

@milyin @DenisBiryukov91 what do you think?

yellowhatter, 2024-11-13:
In opposite to other `as_*` like `as_string`, `as_shm` is used to probe the buffer on shm nature, and unsuccessful result is not as hard error as for orther APIs