eclipse-zenoh/zenoh-c #874: use "get" API for multi-topic

owny990312, 2024-12-23:
### Describe the feature

When implementing an RPC system, I encountered performance issues with the 'get' and 'queueable' APIs. The main problem is that the 'get' API recreates connections for each request, which leads to significant latency overhead, especially in multi-topic scenarios. This connection recreation becomes a bottleneck when handling multiple topics simultaneously.

DenisBiryukov91, 2025-01-06:
Hello @owny990312. Could you try using [Querier](https://zenoh-c.readthedocs.io/en/latest/api.html#_CPPv417z_owned_querier_t) and let us know if it improves performance ?

DenisBiryukov91, 2025-01-06:
Querier to get is a bit like Publisher to put. When you first declare querier, zenoh will perform certain protocol-level optimizations to improve performance of subsequent z_querier_get calls (+ allow checking for matching queryables), that is why I expect it to have an impact on performance in your case.

closed 2025-01-21 by DenisBiryukov91