eclipse-zenoh/zenoh-c #904: [Bug] put vs sub broken in v1.2.0

pdematteis, 2025-02-04:
### Describe the bug

Values sent with a session put to a subscriber have malformed key-expression when using mapped keys: the leading / is missing.
This was not happening in v1.1.1 and does not happen if a publisher is used instead of the session put.
Looking into the network packet with Wireshark, I found that publisher put sends the key with the leading /, while session put does not.

This is a sample packet sent by a publisher
```
0000   02 00 00 00 45 00 00 52 5c f9 40 00 80 06 00 00   ....E..R\.@.....
0010   7f 00 00 01 7f 00 00 01 c5 de 1d 17 93 19 76 69   ..............vi
0020   b3 37 95 f7 50 18 00 fe 80 9c 00 00 28 00 25 ed   .7..P.......(.%.
0030   9c 92 3a 3d 01 0b 2f 6d 61 73 74 65 72 2f 6c 65   ..:=../master/le
0040   64 41 08 12 5b 20 20 20 30 5d 20 50 75 62 20 66   dA..[   0] Pub f
0050   72 6f 6d 20 43 21                                 rom C!
```

This is a packet sent by put, be aware of the missing / in the key
```
0000   02 00 00 00 45 00 00 49 68 b5 40 00 80 06 00 00   ....E..Ih.@.....
0010   7f 00 00 01 7f 00 00 01 c5 f0 1d 17 6c f4 ad 16   ............l...
0020   7f b6 22 b9 50 18 00 fe ff 98 00 00 1f 00 25 84   ..".P.........%.
0030   ee e7 25 3d 01 0a 6d 61 73 74 65 72 2f 6c 65 64   ..%=..master/led
0040   01 0b 50 75 74 20 66 72 6f 6d 20 43 21            ..Put from C!
```

### To reproduce

1. Run a subscriber with `z_sub.exe -k ft/** -l tcp/0.0.0.0:7447`
2. Run a publisher with `z_pub.exe -k ft/master/led -e tcp/127.0.0.1:7447`: the subscriber does receive values
3. Stop the publisher, then run a `z_put.exe -k ft/master/led -e tcp/127.0.0.1:7447`: the subscriber does not receive the value

### System info

Zenoh commit 15d56e1fb44eead6611cf847f503ef030aee5287

DenisBiryukov91, 2025-02-04:
A regression was introduced by Zenoh-rust 1.2.0, the issue is identified and resolved: https://github.com/eclipse-zenoh/zenoh/issues/1755. We will likely make a patch asap.

DenisBiryukov91, 2025-02-05:
The issue is resolved in 1.2.1

closed 2025-02-05 by DenisBiryukov91