eclipse-zenoh/zenoh-c #944: [Bug] Memory leak in Session and SHM publisher destructions

F-Barbieri, 2025-03-10:
### Describe the bug

I ran the following two snippets and the RAM used by the process increased time by time:

```
#include "zenoh.h"
#include <iostream>

int main(int argc, char** argv)
{
    while(true) 
    {
            z_owned_config_t config;
            z_config_default(&config);

            z_owned_session_t s;
            if (z_open(&s, z_move(config), NULL) < 0) {
                std::cout << "Unable to open session!" << std::endl;
            }

            z_drop(z_move(s));
    }
    return 0;
}
```

```
#include "zenoh.h"
#include <iostream>

int main(int argc, char** argv) {
    {
        z_owned_config_t config;
        z_config_default(&config);

        z_owned_session_t s;
        if (z_open(&s, z_move(config), NULL) < 0) {
            std::cout << "Unable to open session!" << std::endl;
            return 1;
        }

        std::cout << "Created session" << std::endl;

        while(true) {
            z_owned_publisher_t pub;

            z_alloc_alignment_t alignment = { 0 };
            z_owned_memory_layout_t layout;

            z_owned_shm_provider_t provider;

            z_view_keyexpr_t ke;
            z_view_keyexpr_from_str(&ke, "Hello");

            z_publisher_options_t pubOptions;
            z_publisher_options_default(&pubOptions);
            pubOptions.reliability = Z_RELIABILITY_BEST_EFFORT;
            pubOptions.congestion_control = Z_CONGESTION_CONTROL_DROP;

            if (z_declare_publisher(z_loan(s), &pub, z_loan(ke), &pubOptions) < 0) {
                printf("Unable to declare Publisher for key expression!\n");
                return 1;
            }

            z_memory_layout_new(&layout, 512, alignment);
            z_posix_shm_provider_new(&provider, z_loan(layout));

            z_drop(z_move(pub));
            z_drop(z_move(provider));
            z_drop(z_move(layout));
        }
        z_drop(z_move(s));
    }
    return 0;
}
```

What went wrong?

### To reproduce

Copy one of the two snippet into a file
Compile the source file customizing the two compile commands provided below
Run the executable
See the RAM usage by _System Monitor_ system utility

### System info

Platform: 
NVIDIA Jetson AGX Orin devkit 64GB (ARMv8)
Jetpack 6.2
Ubuntu 22.04.5 LTS

g++ version: 11.4.0

Compile command:
"g++" -W"switch" -W"no-deprecated-declarations" -W"empty-body" -W"conversion" -W"return-type" -W"parentheses" -W"no-pointer-sign" -W"no-format" -W"uninitialized" -W"unreachable-code" -W"unused-function" -W"unused-value" -W"unused-variable" -std=c++17 -Wall -fno-strict-aliasing -I "/home/<user>/Zenohc_1.2.1/include" -g1 "g++" -O3 -fthreadsafe-statics -D "NDEBUG" -W"switch" -W"no-deprecated-declarations" -W"empty-body" -W"conversion" -W"return-type" -W"parentheses" -W"no-format" -W"uninitialized" -W"unreachable-code" -W"unused-function" -W"unused-value" -W"unused-variable" -frtti -fomit-frame-pointer -std=c11 -fexceptions "1" -o "/home/<user>/projects/Subscriber/obj/ARM64/Release/HelloWorld_main.o" 

Linking command:
-o"/home/<user>/projects/Subscriber/bin/ARM64/Release/Subscriber.out" "3600000" -Wl,-z,relro -Wl,-z,noexecstack -Wl,--no-undefined "g++" -l"zenohc" -l"pthread" -Wl,-L"/home/e80/Zenohc_1.2.1/lib" -Wl,-z,now

DenisBiryukov91, 2025-03-31:
I was not able to reproduce a leak on the first example, I indeed confirm a leak for the second one (with shm).

yellowhatter, 2025-04-03:
Fixed [here](https://github.com/eclipse-zenoh/zenoh/pull/1867)
Need to be propagated to zenoh-c

yellowhatter, 2025-04-03:
Fix contained now in zenoh-c 67e3d8a3b0437f1555641ff29b4e89d4f7bb9dbb

@F-Barbieri you can use latest zenoh-c main branch to see the leak fixed

closed 2025-04-03 by yellowhatter