eclipse-zenoh/zenoh-c #945: [Bug] trying to build zenoh-c using yocto linux fails

ercjethava, 2025-03-12:
### Describe the bug

I am trying to build [zenoh-c ](https://github.com/eclipse-zenoh/zenoh-c) project using yocto.
I am seeing an issue related to cargo not able to pass -Clinker flag properly to rustc compiler when  its trying to build libc crate because of that not able to build zenoh-c using yocto for target architecture which is arm

-> I have tried recipe mentioned into reproduce steps and by manually building one create one by one figure out that issue only occurs when rustc trying to build libc package as defined into cargo.lock file

-> after further investigation found that **linker flag(-Clinker)** provided by RUSTFLAGS doest reflected on rustc compiler when its trying to build libc package for other packages able to verify that rustc compiler able to build without any issue and can see Clinker flag as part of rustc compiler
```
do_compile() {
    export RUSTFLAGS="-Clinker=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc \
    -Car=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ar"
    cd ${S}
    cargo build -p libc --target aarch64-unknown-linux-gnu --verbose 
}

Log data follows:
| DEBUG: Executing shell function do_compile
|    Compiling libc v0.2.158
|      Running `rustc --crate-name build_script_build /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/cargo_home/registry/src/index.crates.io-6f17d22bba15001f/libc-0.2.158/build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="extra_traits"' --cfg 'feature="std"' -C metadata=f471e3c994e9ad30 -C extra-filename=-f471e3c994e9ad30 --out-dir /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/target/debug/build/libc-f471e3c994e9ad30 -L dependency=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/target/debug/deps --cap-lints allow`
| error: linker `cc` not found
|   |
|   = note: No such file or directory (os error 2)

do_compile() {
    export RUSTFLAGS="-Clinker=/usr/bin/cc -Car=${AR}"
    cd ${S}
    cargo build -p allocator-api2 --target aarch64-unknown-linux-gnu --verbose 
}

DEBUG: Executing shell function do_compile
       Dirty allocator-api2 v0.2.18: the rustflags changed
   Compiling allocator-api2 v0.2.18
     Running `rustc --crate-name allocator_api2 --edition=2018 /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/cargo_home/registry/src/index.crates.io-6f17d22bba15001f/allocator-api2-0.2.18/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="alloc"' -C metadata=494c669341fb115e -C extra-filename=-494c669341fb115e --out-dir /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/target/aarch64-unknown-linux-gnu/debug/deps --target aarch64-unknown-linux-gnu -L dependency=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/target/aarch64-unknown-linux-gnu/debug/deps -L dependency=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/target/debug/deps --cap-lints allow -Clinker=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -Car=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ar`
    Finished dev [unoptimized + debuginfo] target(s) in 0.40s
DEBUG: Shell function do_compile finished

```

### To reproduce

-> Add this recipe zenoh-c-lib.bb into yocto: then run it

Step1: Add this into yocto
```
SUMMARY = "Zenoh C Library"
DESCRIPTION = "A C implementation of the Zenoh protocol."
HOMEPAGE = "https://github.com/eclipse-zenoh/zenoh-c"
LICENSE = "EPL-2.0 & Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENSE;md5=530d837aca648e45704db71dedff39c4"

SRC_URI = "git://github.com/eclipse-zenoh/zenoh-c.git;branch=main;protocol=https"
SRCREV = "af53af345cf7263b13954b4553b654c1d9c5cfb5"

PV = "1.0+git${SRCPV}"
S = "${WORKDIR}/git"

DEPENDS += "cmake ninja"

inherit cmake pkgconfig cargo

do_configure() {
    cmake ${S} \
        -DCMAKE_BUILD_TYPE=Release \
        -DCMAKE_SYSROOT="${STAGING_DIR_TARGET}" \
        -DCMAKE_C_COMPILER="${TARGET_SYS}-gcc" \
        -DCMAKE_CXX_COMPILER="${TARGET_SYS}-g++"
}

do_compile() {
[... 114 lines cut ...]

user@ubuntu:/mnt/devkit/build_vm2-la$ bitbake -c devshell zenoh-lib

user@ubuntu:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/git# which aarch64-poky-linux-g++
/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++

user@ubuntu:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/git# which aarch64-poky-linux-gcc
/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc

user@ubuntu:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/git# which aarch64-poky-linux-ar
/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ar

user@ubuntu:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/git# which cc
/usr/bin/cc

```

### System info

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal

DenisBiryukov91, 2025-03-12:
Ok, it seems I understood. 
Unlike allocator-api which builds only lib.rs for TARGET, zenoh-c uses build.rs script to drive the build process.
The build.rs script is compiled FOR HOST and executed by HOST. Then zenoh-c lib.rs is compiled for TARGET. 
I believe specifying linker in RUSTFLAGS, forces cargo to use the same linker for build.rs (HOST) and lib.rs (TARGET) which obviously fails. I think the solution is not to use RUSTFLAGS for specifying linker. You only need to specify --target YOUR_TARGET and --config=target.aarch64-unknown-linux-gnu.linker="\PATH_TO_YOUR_LINKER"\ (quotes seem to be necessary, since cargo expects toml KEY=VALUE format) when invoking cargo.

ercjethava, 2025-03-12:
@DenisBiryukov91 I have tried that same issue :

```
do_compile() {
    cd ${S}
    cargo build --target aarch64-unknown-linux-gnu \
    --config target.aarch64-unknown-linux-gnu.linker=\"/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc\"
}

| error: linker `cc` not found
|   |
|   = note: No such file or directory (os error 2)
| 
```

ercjethava, 2025-03-12:
@DenisBiryukov91 I have tried that same issue :

```
do_compile() {
    cd ${S}
    cargo build --target aarch64-unknown-linux-gnu \
    --config target.aarch64-unknown-linux-gnu.linker=\"/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc\"
}
```

Error:
```
|      Running `rustc --crate-name build_script_build --edition=2021 /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/cargo_home/registry/src/index.crates.io-6f17d22bba15001f/ring-0.17.8/build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="alloc"' --cfg 'feature="default"' --cfg 'feature="dev_urandom_fallback"' -C metadata=a08e3b308d8fe5dd -C extra-filename=-a08e3b308d8fe5dd --out-dir /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/target/debug/build/ring-a08e3b308d8fe5dd -L dependency=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/target/debug/deps --extern cc=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/target/debug/deps/libcc-e05014aa32ff1f08.rlib --cap-lints allow`
|  | error: linker `cc` not found
|   |
|   = note: No such file or directory (os error 2)
```

DenisBiryukov91, 2025-03-12:
I think just making a cmake toolchain file similar to the one we have in ci/toolchains without touching RUSTFLAGS, should be easier.

ercjethava, 2025-03-12:
@DenisBiryukov91 yes that is from yocto recipe envoronment result

ercjethava, 2025-03-12:
@DenisBiryukov91 have also tried using .cmake file

```
do_configure() {
    cmake ${S} \
        -DCMAKE_TOOLCHAIN_FILE=${S}/TC-aarch64-unknown-linux-gnu.cmake
}
```

Error:
```
| error: linker `cc` not found
|   |
|   = note: No such file or directory (os error 2)
```

DenisBiryukov91, 2025-03-12:
Can you show the full error and list of commands that you run  ?

Jarsop, 2025-03-12:
Copy/paste from Discord z-support channel for visibility:

I pushed working zenoh-c recipe in my repo: https://github.com/Jarsop/meta-zenoh.git (1.2.1 based) but my layer is constrained to scarthgap. You can try to copy paste the recipe files in your own meta layer and try to build it.

Some feedback about your recipe:

* In order to have reproducible build next Yocto version disable network connection for other tasks than `do_fetch`. The right way is to declare any crates dependencies in `SCR_URI` to use bitbake fetcher.
* I strongly recommend to use `cmake` build provided by the official repo as it define an install target so you will have C headers, cmake additional files and pkgconfig file installed in your dev package (necessary to build with cmake in Yocto and if you build a SDK). If not you have to install it manually and potentially customize it.
* Your DEPENDS definition is wrong because it causes a dependency on `cmake` and `ninja` for your target and not for your host (that you want because you build on your host), the right way is `cmake-native` and `ninja-native` but it automatically included by `cmake` bbclass.

ercjethava, 2025-03-12:
> Also do you have a rust toolchain for x86_64-linux (which seems to be your host), can you ensure that calling cargo/ rustc from yocto environment whithout any arguments invokes them for x86_64-linux toolchain ?

This is my **target.cmake** file:
```
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_SYSTEM_PROCESSOR aarch64)
set(DEBARCH arm64)
set(ZENOHC_CUSTOM_TARGET aarch64-unknown-linux-gnu)
set(ZENOHC_CARGO_FLAGS "--config=target.aarch64-unknown-linux-gnu.linker=\"aarch64-linux-gnu-gcc\"")
set(CMAKE_C_COMPILER /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc)
set(CMAKE_CXX_COMPILER /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++)
```

**recipe:**
```
SUMMARY = "Zenoh C Library"
DESCRIPTION = "A C implementation of the Zenoh protocol."
HOMEPAGE = "https://github.com/eclipse-zenoh/zenoh-c"
LICENSE = "EPL-2.0 & Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENSE;md5=530d837aca648e45704db71dedff39c4"

SRC_URI = "git://github.com/eclipse-zenoh/zenoh-c.git;branch=main;protocol=https"
SRCREV = "af53af345cf7263b13954b4553b654c1d9c5cfb5"

PV = "1.0+git${SRCPV}"
S = "${WORKDIR}/git"

DEPENDS += "cmake"

inherit cmake pkgconfig cargo

do_configure() {
    cmake ${S} \
        -DCMAKE_TOOLCHAIN_FILE=${S}/target.cmake
}
```

**Error:**
bitbake <recipe>
```
|      Running `rustc --crate-name build_script_build --edition=2021 /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/cargo_home/git/checkouts/zenoh-cc237f2570fab813/2311334/commons/zenoh-macros/build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debug-assertions=off -C metadata=21a08f1d7ec22089 -C extra-filename=-21a08f1d7ec22089 --out-dir /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/target/release/build/zenoh-macros-21a08f1d7ec22089 -L dependency=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/target/release/deps --cap-lints allow`
| error: linker `cc` not found
|   |
|   = note: No such file or directory (os error 2)
| 

```

**Host PC:**
```
root@user:/mnt/devkit/build_vm2-la$ which cc
/usr/bin/cc

root@user:/mnt/devkit/build_vm2-la$ which rustc
/home/v920/.cargo/bin/rustc

root@user:/mnt/devkit/build_vm2-la$ which cargo
/home/v920/.cargo/bin/cargo

```

DenisBiryukov91, 2025-03-12:
I think target.aarch64-unknown-linux-gnu.linker show point to the same linker as the one being used for c compiler.

Another issue, when you showed which cargo/rust on discord the result was pointing to /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin and not  /home/v920/.cargo/bin/, which makes me thing that the issue is in the way you invoke cargo, can you verify that rust installation in  /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin contains the toolchain necessary for the host (i.e x86_64-linux)?

ercjethava, 2025-03-12:
@DenisBiryukov91 
yes it does have: 

```
root@user:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin$ ls
aarch64-poky-linux  c_rehash   colrm         flex++         infocmp     look      lzgrep           ncursesw5-config  pkg-config-native  rust-gdb      taskset    wall                            xzegrep
aclocal             cal        column        flex.real      infotocap   lsblk     lzless           ncursesw6-config  prlimit            rust-gdbgui   tic        wdctl                           xzfgrep
aclocal-1.16        captest    cpack         flock          ionice      lscpu     lzma             netcap            pscap              rust-lldb     toe        whereis                         xzgrep
attr                captoinfo  crossscripts  gdbm_dump      ipcmk       lsipc     lzmadec          ninja             pseudo             rustc         tput       write                           xzless
autoconf            cargo      ctest         gdbm_load      ipcrm       lslocks   lzmainfo         nsenter           pseudodb           script        tset       x86_64                          xzmore
autoheader          ccmake     curl          gdbmtool       ipcs        lslogins  lzmore           openssl           pseudolog          scriptlive    ul         x86_64-linux-libtool
autom4te            chmem      curl-config   getfattr       isosize     lsmem     m4               openssl.real      python3-native     scriptreplay  uname26    x86_64-pc-linux-gnu-pkg-config
automake            choom      eject         gnu-configize  last        lsns      mcookie          patch             quilt              setarch       unlzma     xmlwf
automake-1.16       chrt       fallocate     guards         lastb       lzcat     mesg             pcre2-config      re2c               setfattr      unshare    xz
autoreconf          clear      filecap       hardlink       libtoolize  lzcmp     namei            pcre2grep         rename             setsid        unxz       xzcat
autoscan            cmake      fincore       hexdump        linux32     lzdiff    nativepython3    pcre2test         renice             setterm       utmpdump   xzcmp
autoupdate          col        findmnt       i386           linux64     lzegrep   ncurses5-config  pkg-config        reset              sqlite3       uuidgen    xzdec
bzip2-native        colcrt     flex          ifnames        logger      lzfgrep   ncurses6-config  pkg-config-esdk   rev                tabs          uuidparse  xzdiff
v920@v920-jeremy-richards-3:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin$ 
```

DenisBiryukov91, 2025-03-12:
> Host PC installed rust which I dont think yocto is using:

That seems to be a problem, since there is no `cc` under /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin, so the yocto's rustc can not find it. Same question about RUST x86_64-linux toolchain, under the same root ...

ercjethava, 2025-03-12:
@DenisBiryukov91 

**1. cc linker issue:**
-> so you mean I need to point **native cc compiler (/usr/bin/cc)** to /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin then it will resolve linker cc not found issue

**2.  rust  x86_64-linux toolchain**

-> what do you mean by x86_64-linux toolchain  ?
-> Do you mean host system installed **rustc(/home/v920/.cargo/bin/rustc )** and **cargo(/home/v920/.cargo/bin/cargo)**  should be under :mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin 

-> So then you mean build has to use native rustc , cargo instead of yocto generated rustc, cargo which is under :

```
root@user:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin
```

as you can see this path already has rustc and cargo which is built by yocto:

```
root@user:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin$ ./rustc --version
rustc 1.72.0 (5680fa18f 2023-08-23) (built from a source tarball)
root@user:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin$ 
root@user:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin$ ./cargo --version
cargo 1.72.0
root@user:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin$ 

```

As at the moment you can see bitbake using yocto generated rustc and cargo: which is under:
 /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/

```
root@v920-jeremy-richards-3:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin# ls -all | grep rustc
-rwxr-xr-x    1 1000 1000    16776 Mar 12 02:21 rustc
root@v920-jeremy-richards-3:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin# ls -all | grep cargo
-rwxr-xr-x    1 1000 1000 19935576 Mar 12 02:23 cargo
root@v920-jeremy-richards-3:/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib/1.0-r0/recipe-sysroot-native/usr/bin# 
```

Build Logs:
```
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: which rustc: /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/rustc
| NOTE: rustc --version rustc 1.75.0 (82e1608df 2023-12-21)
| NOTE: which cargo: /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: cargo --version cargo 1.75.0 (1d8b05cdd 2023-11-20)
| NOTE: cargo build --verbose --manifest-path /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/git/Cargo.toml --target=aarch64-unknown-linux-gnu --release
|    Compiling proc-macro2 v1.0.86

|      Running `rustc --crate-name build_script_build --edition=2021 /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/cargo_home/registry/src/index.crates.io-6f17d22bba15001f/rustls-0.23.23/build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debug-assertions=off --cfg 'feature="log"' --cfg 'feature="logging"' --cfg 'feature="ring"' --cfg 'feature="std"' --cfg 'feature="tls12"' -C metadata=76c9de69d605d4c1 -C extra-filename=-76c9de69d605d4c1 --out-dir /mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/target/release/build/rustls-76c9de69d605d4c1 -L dependency=/mnt/devkit/build_vm2-la/tmp/work/aarch64-poky-linux/zenoh-lib2/1.0+gitAUTOINC+af53af345c-r0/target/release/deps --cap-lints allow`
| error: linker `cc` not found

```

Jarsop, 2025-03-12:
All toolchain executables are located in ${bindir}/aarch64-poky-linux folder. But Yocto already create your cmake toolchain file thanks to cmake bbclass (but your configure function has been overridden).
It's not properly integrated in Yocto but FYI Yocto use gcc or ld (aarch64-poky-linux-gcc and aarch64-poky-linux-ld in your case)

ercjethava, 2025-03-12:
@Jarsop  so in my case any idea how can I fix to setup environment properly as this is my recipe : as I am just configuring cmake as per target.cmake file which is based on target.cmake and recipe file I am using as mentioned here:
https://github.com/eclipse-zenoh/zenoh-c/issues/945#issuecomment-2718513699

Jarsop, 2025-03-12:
It might be tricky as Yocto is not designed for this kind of integration. I recommend to use the default cmake `do_configure` and use `EXTRA_OECMAKE` variable to pass any additional zenoh flag. Regarding the compile task you can try to call `oe_cargo_fix_env` defined in `cargo_common.bbclass` and then the default cmake `do_compile` command. As example you can look my recipe in the repo. Do not hesitate if I'm not clear, sorry I write from my cell phone.

Jarsop, 2025-03-12:
Another idea: try to use `aarch64-poky-linux-gcc` as linker for cargo

ercjethava, 2025-03-12:
@Jarsop  that's what I have tried by passing linker information via export RUSTFLAGS as metnioned into zenoh-c project readme file, however issue relies specific when cargo is trying to build libc rust crates as its not able to find specific cc linker , all other crates build perfectly which I have verified by building one package at a time using cargo -p <package_name>  which I already posted it as a initial issue here:
https://github.com/eclipse-zenoh/zenoh-c/issues/945#issue-2912199504

-> This is example from zenoh-c README.md file same I followed: I am using RUSTFLAGS to pass linker as aarch64-poky-linux-gcc and ar as aarch64-poky-linux-ar  which is failing when using same on yocto environment as somehow only for libc crates cargo somehow using cc linker instead of aarch64-poky-linux-gcc and its failing to build libc crates and so whole compilation is failing if somehow I can fix this specific libc rust crates compilation issue then it will pass 
```
export RUSTFLAGS="-Clinker=x86_64-w64-mingw32-gcc -Car=x86_64-w64-mingw32-ar"
mkdir -p build && cd build
cmake ../zenoh-c  -DCMAKE_SYSTEM_NAME="Windows" -DCMAKE_C_COMPILER="x86_64-w64-mingw32-gcc" -DCMAKE_CXX_COMPILER="x86_64-w64-mingw32-g++" -DCMAKE_SYSTEM_PROCESSOR="x86_64" -DZENOHC_CARGO_CHANNEL="+nightly" -DZENOHC_CARGO_FLAGS="-Zbuild-std=std,panic_abort" -DZENOHC_CUSTOM_TARGET="x86_64-pc-windows-gnu" -DCMAKE_INSTALL_PREFIX="../x86_64-pc-windows-gnu/stage"
cmake --build . --target install
```

-> So what I am  thinking is that there should be two different macro defined using which when cargo trying to build libc crate we need to force rustc compiler to use linker as cc compiler and for all other crates its able to build successful using provided linker path using export RUSTFLAGS as arm cross compiler

Jarsop, 2025-03-15:
This issue is not related to `zenoh-c` but to Yocto integration, I think this can be closed 😊

ercjethava, 2025-03-17:
@Jarsop  yes I am closing this as already able to build on yocto using help from @Jarsop

closed 2025-03-17 by ercjethava